QIDITECH/QIDI_PLUS4 #3: Link QR Code Doesn't Work, Stuck Loading

keiz-7, 2024-09-30:
The QR Code for Linking to the app does not work, and it is stuck loading without a QR code popping up. I've reset the firmware, clipper, and factory settings, and also tried removing and re-adding the wifi network, but none of that is working.

whb0514, 2024-09-30:
1.Firstly, confirm whether the server selection is AWS, and then select again.
2. Confirm that your network is connected to the internet.
3. Clicking on restore factory settings should be able to solve the problem.
4. Please confirm if the router has set up any security policies.
If none of the above is possible, please export the log file and send it to us for after-sales inspection.

keiz-7, 2024-10-03:
I was able to connect it to Qidi link for only about 20 seconds after turning my plus 4 on, however now the printer always says it's offline, sometimes it will connect for about 15 to 20 seconds if I turn the printer off and back on. The connection says AWS in the app, however if I click server in the network menu on the plus 4, nothing is listed.  It is connected to the internet, I can see it on my router's list of connected devices. I clicked restore factory settings that didn't fix the issue. My router's only security policies are just password on the wireless. I have issues as well connecting the printer to Qidi Studio as well, I get connection timeouts and cannot connect it to Qidi studio. 

I have the log files exported. Where should I send them?

QIDITECH, 2024-10-04:
I think this is your network problem, he can not stably connect to the AWS server, please try to change the router or use the mobile phone hotspot to see if there is any improvement.

PKayUK, 2024-10-06:
I have/had the same issues. Using ethernet  between the printer and hub seems to have fixed it. I think it's an issue with using dual band wifi (2.4/5GHz)  Printer on 2.4 and PC on 5.  Never had this issue with other devices on my  home lan and don't want to split my network. Wifi hardware supplied by UK's largest broadband provider (around 33% of market) so I'm reluctant to to believe their hardware has a bug.

closed 2024-11-20 by QIDITECH