QIDITECH/QIDI_PLUS4 #4: Support e-mail bounces

pcwilhelm, 2024-10-06:
If you need to contact QIDI support you may be disappointed. My simple e-mail to Plus4support@qidi3d.com bounced back as spam.  HUH? Seems kind of rude at the moment. Hopefully I’m not stuck with an $800 boat anchor. I’ll report back.

QIDITECH, 2024-10-07:
Sorry, you can try to contact us at another email, Plus4Ams@qidi3d.com, or let me know what help you need.

Tyler-DurdenRUL1, 2024-10-07:
um, I don't know what is going on with his issue, but I cant update my brand new plus4 printer. I set everything up and ran through the setup like I've done with the other 6 Qidi printers I own. I downloaded the newest firmware update and put in on the thumb drive. put it in the printer and it wont fully update. there is an error message on the screen, and from the error message it says the UI updated but the "soc" is not correctly updating. so it has conflicting versions and displays an error message. ever since Qidi did one of the updates about 5 months ago, this happens to all my printers now when trying to update them, every single time! my x max3 printers never did this, until 5 months ago. both my q1 pro's have the same issue. usually if I copy and remove the backup file portion from the thumb drive temporarily, the update with go through. no matter what I try with my plus4, nothing is working! I haven't even got to print a single damn thing on it yet. I always update to the newest version as soon as I unpack all of my Qidi printers. I am not sure if that's the same issue that he is having, but I've had my plus4 for 3 days now and I cant even use it yet because of this bs error. I am about to rip my hair out.

pcwilhelm, 2024-10-08:
> Sorry, you can try to contact us at another email, [Plus4Ams@qidi3d.com](mailto:Plus4Ams@qidi3d.com), [karl@qd3dprinter.com](mailto:karl@qd3dprinter.com), or let me know what help you need.

I tried to contact you 24 hours ago at the Plus4Ams@qidi3d,com as recommended. no response yet.

whb0514, 2024-10-08:
> 嗯，我不知道他的问题是怎么回事，但我无法更新我全新的plus4打印机。我设置了所有的东西，并像使用我拥有的其他6台蒂奇打印机一样运行了设置。我下载了最新的固件更新，放在u盘里。把它放进打印机，它不会完全更新。屏幕上出现一条错误消息，错误消息显示用户界面已更新，但“soc”没有正确更新。因此它有冲突的版本并显示一条错误消息。自从蒂奇5个月前做了一次更新后，我所有的打印机每次更新时都会发生这种情况！我的x max3打印机从来没有这样做过，直到5个月前。我的两个q1 pro都有相同的问题。通常，如果我暂时从u盘中复制并删除备份文件，更新就会完成。无论我用我的plus4做什么，都没有用！我甚至还没在上面印任何东西。我总是一打开我所有的蒂奇打印机就更新到最新版本。我不确定这是不是和他遇到的问题一样，但是我已经用plus4三天了，因为这个bs错误，我甚至还不能使用它。我要把我的头发拔掉了。

The issue seems to be caused by an operational error on your part, resulting in an incomplete update. We need you to provide more detailed information, preferably in the form of images, such as screenshots of the update content on the USB drive, the UI interface’s homepage notification about an incorrect version, and the screenshots of the update detection page.

whb0514, 2024-10-08:
> um, I don't know what is going on with his issue, but I cant update my brand new plus4 printer. I set everything up and ran through the setup like I've done with the other 6 Qidi printers I own. I [...]

The issue seems to be caused by an operational error on your part, resulting in an incomplete update. We need you to provide more detailed information, preferably in the form of images, such as screenshots of the update content on the USB drive, the UI interface’s homepage notification about an incorrect version, and the screenshots of the update detection page.

QIDITECH, 2024-10-09:
@pcwilhelm Sorry, we have checked the mailbox and did not receive your letter. Maybe you need to change the mailbox to send the message, or tell me your mailbox, we will take the initiative to contact you.

closed 2024-10-25 by QIDITECH