QIDITECH/QIDI_PLUS4 #5: [Improvement] Display Filament Load Screen When Filament Is Inserted

spooknik, 2024-10-09:
Hello, I am a happy owner of a Plus 4.

It would be a nice quality of life feature to have the screen show the filament load menu when filament is inserted past the filament switch (tangle sensor). It saves the user a click and is just convenient. 

For example: 

```
[filament_switch_sensor fila]
pause_on_runout: True
runout_gcode:
    M118 Filament tangle detected
insert_gcode: load_screen_macro # Call the load screen menu in Xindi. 
event_delay: 3.0
pause_delay: 0.5
switch_pin:U_1:PC3
```

whb0514, 2024-10-09:
> Hello, I am a happy owner of a Plus 4.
>
> It would be a nice quality of life feature to have the screen show the filament load menu when filament is inserted past the filament switch (tangle sensor). [...]

Does it mean that the system automatically feeds a certain amount of material after detecting the consumables? You can first add this custom setting, and we will consider adding it later.

spooknik, 2024-10-09:
> Does it mean that the system automatically feeds a certain amount of material after detecting the consumables? You can first add this custom setting, and we will consider adding it later.

Yes this is also a good solution and a nice feature. 

However I meant if this screen could appear after detecting the consumables.  So the user does not need to press 'Load' in UI, it is just done for them. 
[image]

QIDITECH, 2024-10-10:
Because the consumables detection is located above the extruder gear, it is not very reliable to rely on it alone, and the user still needs to pull the filament to determine whether the feed is successful.