QIDITECH/QIDI_PLUS4 #8: firmware

capnkirk643, 2024-10-11:
qidi plus 4 unable to update and supplied usb isn't recognized, printer shuts off when i heat the bed. nothing works

spooknik, 2024-10-11:
Hello! We had a similar issue with a few users on the Qidi Discord. Check your power supply is set to the correct voltage for your region.

whb0514, 2024-10-11:
> qidi plus 4 unable to update and supplied usb isn't recognized, printer shuts off when i heat the bed. nothing works

Regarding the USB update issue, please contact the after-sales service, they will troubleshoot the problem. For the issue of hot bed heating shutdown, please first check if the voltage of the switch power supply is consistent with your region.

capnkirk643, 2024-10-11:
Will do and i havent seen anything about the power supply regarding switching it to 120 or 220. Where might that switch be?

spooknik, 2024-10-11:
> Will do and i havent seen anything about the power supply regarding switching it to 120 or 220. Where might that switch be?

It's on the bottom of the printer. 

See here > https://www.youtube.com/live/TsiXYSRfq74?si=NfUzHAjOu7ohpjbX&t=5416

capnkirk643, 2024-10-11:
Thank you very much spooknik, that was exactly the problem

capnkirk643, 2024-10-26:
check power supply located under printer

closed 2024-10-26 by capnkirk643