QIDITECH/QIDI_PLUS4 #12: Pre 1.4 firmware offline update failure using firmware v1.4.2!!!

Assistedevolution, 2024-10-20:
Doing a first offline update on a new machine, with pre 1.4 firmware. going to latest 1.4.2 release

The printer found the update file on the USB. It got to the following screen (see image below) and does not progress any further. It remains at 0%. (see screen shot).

After 1 hour tried to power cycle the printer but on reboot it just returns to this screen with progress 0%. 

There seems to be no way to get to any other screen. The system is totally unresponsive.

Power cycling, either with or without the USB installed does nothing... it just boots and immediately goes to this screen.

Any help would be appreciated.
[image]

Assistedevolution, 2024-10-21:
Note left for a further 8 hours and still showing zero progress. This morning attempted the firmware mksscreen.recovery process as documented in the wiki in case that might be an issues but machine is still stuck at the same screen and is effectively unresponsive.

whb0514, 2024-10-21:
Please contact our after-sales service to replace a screen for you.

Assistedevolution, 2024-10-22:
Ok issue resolved. Thanks to Qidi support. 
Problem turned out to be a corrupt RAR extraction, probably our from our side. Qidi sent a pre extracted UI file renamed to mksscreen.recovery which solved this issue, and updated the UI. 
The reason this recovery process had not worked for us before is because our UI file was corrupt. Worth checking if you have a similar problem

closed 2024-10-22 by Assistedevolution