QIDITECH/QIDI_PLUS4 #14: Plus 4 arrived without a screen

Wayfst, 2024-10-23:
While unboxing I found an empty box for the screen. Email several times no response, anyone have an email that works?

Rizadium, 2024-10-23:
I used Plus4support@qidi3d.com

QIDITECH, 2024-10-24:
You can try to contact us at email, [Plus4support@qidi3d.com](mailto:Plus4support@qidi3d.com), [Plus4Ams@qidi3d.com](mailto:Plus4Ams@qidi3d.com), [karl@qd3dprinter.com](mailto:karl@qd3dprinter.com).

Wayfst, 2024-10-24:
> CAUTION: Do not click links or open attachments if you do not recognize the sender.

I did

Sent from my Verizon, Samsung Galaxy smartphone

QIDITECH, 2024-10-25:
If you still haven't received a reply, please tell me your email address, we will take the initiative to contact you.

closed 2024-11-20 by QIDITECH