QIDITECH/QIDI_PLUS4 #15: SSR Chamber heater meltdown

FxWoody, 2024-10-24:
The SSR H-2JRMK1.0 are not load compatible with North America 115V system and the  Choke Inductor meltdown from overheating when applying electrical load on chamber heater.
Issue is starting to spred widely threw many owners using the heater. 

This issue is not beeing involved with the "low bed plate choke" issue that create a overheat shut off issue.
[https://youtu.be/YLzQhrdXodQ?si=Jilzm60PuIG9dD3I](url)

[image]

67RS427, 2024-10-24:
There is a firmware fix. If you contact them, they will also send you a revised SSR, free of charge.

QIDITECH, 2024-10-25:
We have noticed that the previous firmware settings may cause some users in the U.S. to experience a momentary surge in current. We have released a new firmware update that addresses this issue. Please update to the latest firmware to resolve the situation.
Rest assured, this issue will not cause a fire, as the device will automatically disconnect if the current exceeds safe levels to ensure your safety. If you need any assistance or a replacement part, please feel free to contact our after-sales support team, and we will provide a free replacement.

closed 2024-10-27 by FxWoody