QIDITECH/QIDI_PLUS4 #18: Remove update function in Fluidd

FxWoody, 2024-10-27:
Since the klipper version is modify for the Printer, the Fluidd "update" function should be removed or locked.
I had a notification message in the dashboard to update and did it but ended up "bricking" my printer or causing errors and now printer is not working anymore.
Need info how to fix this :)
[image]
[image]

QIDITECH, 2024-10-28:
We have not enabled this function, please confirm whether it is caused by your operation.

FxWoody, 2024-10-28:
The update function in Fluidd is working from the bottom of the settings page ... i even had a notification that came up in the upper bar. Was showing updates needed so i did :(

QIDITECH, 2024-10-29:
The version of fluidd we use is not very different from the official latest version, if it is only updated fluidd, it should not affect the use. Be careful not to update the moonraker or klipper, as this will render the printer unusable.
Maybe you can find the configuration in the moonraker.cfg file in the configuration to turn it off.
Or you can update the firmware offline to override the original version.

closed 2024-11-11 by FxWoody