QIDITECH/QIDI_PLUS4 #19: Info request/tooltips

WyldCFH, 2024-10-29:
Is there a way to get some expanded info or tooltips in the Fluidd interface to explain what Sound, Beeper and ctlyd switches do?

[image]

QIDITECH, 2024-10-29:
The current version of fluidd does not support this.

closed 2024-10-29 by QIDITECH

WyldCFH, 2024-10-29:
Can this info be added to the Wiki then? 
Or, just put the explanation here so the next person that comes looking for it can find it.

QIDITECH, 2024-10-30:
Sound and Beeper: Used to control a buzzer that will sound an alert at zero and the end of printing
Ctlyd: For the control of piezoelectric plates
Caselight: Used for LED lights inside the case

WyldCFH, 2024-10-30:
Thank you for the explanation.