QIDITECH/QIDI_PLUS4 #20: Print notifications

WyldCFH, 2024-10-29:
I am not receiving notifications in QidiStudio, QidiLink or MobileRaker for print statuses.

I receiving nothing for print finish, print pause for any reason incl. filament runout, print finish or any notifications at all. This has been the case from the very beginning. 
I had hoped the newest firmware would fix it, but it has not

QIDITECH, 2024-10-29:
Sorry, this feature is not currently available.

WyldCFH, 2024-10-29:
Seems like some programming logic that needs to be added. Sub-$200USD printers running Klipper have no issues sending notifications on status changes

At this point, the printer prints amazingly well, but if it weren't out of return period I'd return it, buy a P1S and spend $50 to put a chamber heater in it.

nicwilson58, 2024-11-04:
> Seems like some programming logic that needs to be added. Sub-$200USD printers running Klipper have no issues sending notifications on status changes
>
> At this point, the printer prints amazingly [...]

retun it over a useless notification?  Takes all kinds I suppose.

WyldCFH, 2024-11-04:
> retun it over a useless notification? Takes all kinds I suppose.

So when was the last time you didn't sit beside and monitor your ABS print waiting for it to run out of filament?
If you have alife outside of printing, it is very easy to not monitor your print and not know it paused. If that pause is extended, your print could very well cool too much and shrink, then upon resume it is potentially a bad print.

Add the fast that 120v chamber heaters had their heating capacity neutered to keep them from melting down, and the nozzles were changed from launch, which was only a couple months ago, to help stop heat creep, and they have a special print(shim) to fix the bottom right corner of the door not closing all the way, and "some" have needed to open the machine to pull the wifi adapter and plug into the top so they can connect to their networks, and .....

Sooo, until you know the full story you may want to reserve your judments and/or keep them to yourself. Otherwise, someone like me may respond and make you look like and idiot

nicwilson58, 2024-11-04:
I know the full story,  I have one,   and all I see and hear is moaning

nicwilson58, 2024-11-04:
> 
> Sooo, until you know the full story you may want to reserve your judments and/or keep them to yourself. Otherwise, someone like me may respond and make you look like and idiot

I know the story I have the same printer,  and all I mostly see it moaning.  the chamber has NOT been crippled as even at 4uduty cycle it heats up faster than the bed for ABS.   I calculate required filament first,  and make sure there is enough or what time it wil run out, I also setup my own notifications from Klipper.    I find not issue with the door and no shims needed,  but even so that is a whine,  if the wifi does not connect where it already is improve your network, my router is across the other side of a very big house and i have no issues with it,  or any of my other printers connectiing,  you are moaning for the sake of moaning instead of just getting on with it.

WyldCFH, 2024-11-04:
If all you hear is moaning, maybe you should stay off of the ISSUES board of Github, and all social media. But, I guess,  idiots do not understand that simple, logical, course of action.

But, you decide to come here and make yourself look bad by making a comment, that is of absolute use to anyone or anything, while claiming to know all the issues. So I must make the assumption, that you just wanted to make someone else look bad by claiming "it takes all kinds" while "knowing the story", thereby just making yourself look worse.

Go crawl in your corner somewhere unless you have some value to add or post a new issue

Kammi61871, 2024-11-10:
why qidi dont use klipper api ?
Enabling the API socket[¶](https://www.klipper3d.org/API_Server.html#enabling-the-api-socket)

In order to use the API server, the klippy.py host software must be started with the -a parameter. For example:

`~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer.cfg -a /tmp/klippy_uds -l /tmp/klippy.log
`

nicwilson58, 2024-11-13:
> 
> If all you hear is moaning, maybe you should stay off of the ISSUES board of Github, and all social media. But, I guess, idiots do not understand that simple, logical, course of action.
> 
> But, you decide to come here and make yourself look bad by making a comment, that is of absolutely no use to anyone or anything, while claiming to know all the issues. So I must make the assumption, that you just wanted to make someone else look bad by claiming "it takes all kinds" while "knowing the story", thereby just making yourself look worse.
> 
> Go crawl in your corner somewhere unless you have some value to add or post a new issue

As you moan was also of no use to anyone.  if you dont like it fix it or did you miss the part where I fixed mine?

WyldCFH, 2024-11-13:
> > Go crawl in your corner somewhere unless you have some value to add or post a new issue
> 
> As your moan was also of no use to anyone. if you dont like it fix it or did you miss the part where I fixed mine?

No no, I saw what you claimed. You've provided no proof or process of actul fixes. Where as I, have come to the correct place to request the Manufacturer fix something for EVERYONE. Not just be a troll, like you, and fix for myself only

More ignorance displayed by someone that has no life but frequent boards meant for troubleshooting, communicating with developers and then complain about someone trying to get something fixed

nicwilson58, 2024-11-13:
Grow up and stop complaining.

WyldCFH, 2024-11-13:
Really? That's all you got? "Grow up and stop complaining".

You are the one here complaining, over and over. I came here to try and make things better, as mentioned, multiple times. 

I fail to understand the sheer lack of self realization you have shown in being the exact problem you are claiming I am perpetrating. Maybe it just requires a bit more intelligence than you have.

That said, you can respond a thousand more times with your "quit complaining" "quit moaning" schtick and I won't respond to you again.

And, the simple fact that noone from Qidi has returned, in 2 weeks, and commented anything about the original issue is yet another reason they do not deserve to be supported.

nicwilson58, 2024-11-13:
What part of stop moaning don’t you understand?

tschreiner, 2024-11-14:
Please be respectful. There is no need to be confrontative. You can use Mobileraker for notifications. You just have to install the Mobileraker companion  daemon on the printer. This can be done easily over SSH and kiauh. Takes 3 minutes.

WyldCFH, 2024-11-14:
> Please be respectful. There is no need to be confrontative. You can use Mobileraker for notifications. You just have to install the Mobileraker companion daemon on the printer. This can be done easily over SSH and kiauh. Takes 3 minutes.

Thanks,I have alreaady done this myself.

I still believe this should be added by default. Not every user would know how to go about finding this board, using SSH or have the confidence to fire up kiauh and start installing stuff *if* they could find the info on a general Google search

nicwilson58, 2024-11-24:
The loud resonance I found to be the known issue of the loose lower heat bearing, I fitted the two M3 screws from the back as suggested by Qidi and mine is now mostly silent when doing input shaping tests.