QIDITECH/QIDI_PLUS4 #21: SSH to qidi plus 4

tuannat, 2024-11-01:
How can I access to qidi plus instance through SSH?

67RS427, 2024-11-01:
Any SSH GUI will work. There are two users in the system provided by Makerbase:
root user: root
user2: mks
their origin passwords are the same: makerbase

FxWoody, 2024-11-04:
> Any SSH GUI will work. There are two users in the system provided by Makerbase: root user: root user2: mks their origin passwords are the same: makerbase

root password is not makerbase :(

67RS427, 2024-11-04:
> > Any SSH GUI will work. There are two users in the system provided by Makerbase: root user: root user2: mks their origin passwords are the same: makerbase
> 
> root password is not makerbase :(

It should be... worked for me with user: mks pass: makerbase

FxWoody, 2024-11-04:
yes for the MKS it works but not in ROOT :(

WyldCFH, 2024-11-04:
User mks should give you full root access. 
I don't recall ever seeing a user for Makerbase boards as "root"

Rizadium, 2024-11-06:
> yes for the MKS it works but not in ROOT :(

just reset it like i did.
Log in as mks
type in sudo -i and hit enter
It will ask for the mks password (makerbase)
type in sudo passwd root and enter
Enter in new root password 2x and win!
[image]

QIDITECH, 2024-11-07:
In order to improve the network security of the printer, we only disclose the password of the mks account, and the password of the root account is no longer public.
For users, installing plug-ins or upgrading software can be done using the mks account and does not require root permission.

AMoo-Miki, 2024-11-13:
@QIDITECH would anything stop working if the root password is changed?

FxWoody, 2024-11-13:
Absolutely not unless you make a mess in the root repo, just give you higher access if you want to modify anything. I was asking because i had install a mod that i didn't use but that i could not delete because it needed root access 😉

QIDITECH, 2024-11-14:
Sorry, there seems to be no other way for you to delete these files, but fortunately the printer has enough memory and deleting these files is not necessary.

rogernolan, 2024-11-18:
This password should be documented somewhere better than this github issue and the Reddit comment with the same info. It’s very difficult to track down.