QIDITECH/QIDI_PLUS4 #27: Fix temperature settings screen

lukaszdrukasz, 2024-11-12:
Pls pls fix the layout of the virtual numpad used to set the temperature from the touchscreen. Currently the digits on the screen are organized like this:

1 2 3 4
5 6 7 8
9 0

which is against humanity's muscle memory that expects

1 2 3
4 5 6
7 8 9
&nbsp;&nbsp;&nbsp;0

or

7 8 9
4 5 6
1 2 3
&nbsp;&nbsp;&nbsp;0

I can't count how many times I set the bed to 70 instead of 60 C...

67RS427, 2024-11-12:
That's probably not gonna happen.... Set up a preset to make your life easier. I've never had a problem typing in the wrong numbers and most of the time I do it in the Fluidd interface anyway.

QIDITECH, 2024-11-13:
We tried putting the numbers in four rows, but with the current screen size and layout, it is difficult to achieve a nice and clean layout.
We will continue to experiment with new designs and look forward to updates in future versions.

AMoo-Miki, 2024-11-13:
If 4 rows is a problem, perhaps one of these layouts are more user friendly?
```
0   1   2   3
    4   5   6
    7   8   9
```
or
```
1   2   3
4   5   6
7   8   9   0
```