QIDITECH/QIDI_PLUS4 #28: Klipper update for adaptive pressure advance

67RS427, 2024-11-15:
Can we get an update to Klipper to a version greater than v0.12.0-267 so we can run Adaptive Pressure Advance?

Issue:
"Older versions of Klipper used to stutter when pressure advance was changed while the toolhead was in motion. This has been fixed with the latest Klipper firmware releases. Therefore, make sure your Klipper installation is updated to the latest version before enabling this feature, in order to avoid any adverse quality impacts.

Klipper firmware released after July 11th, 2024 (version greater than approximately v0.12.0-267) contains the above fix and is compatible with adaptive pressure advance. If you are upgrading from an older version, make sure you update both your Klipper installation as well as reflash the printer MCU’s (main board and toolhead board if present)."

https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-pressure-advance

Thanks!

QIDITECH, 2024-11-16:
This may cause some functions to not work properly, such as automatic leveling, chamber heating, etc.

AlAmantea, 2024-11-28:
> This may cause some functions to not work properly, such as automatic leveling, chamber heating, etc.

This is why we are asking you to update it FOR US in the firmware, so we don't break any other functions.

ocwins, 2024-12-30:
we need this too.

fixed pa can't work well with both high and low speed.

erasmus74, 2025-01-04:
This is an issue that is reducing quality of prints. Please fix.

gogsi1, 2025-02-07:
Update the firmware to the current version of klipper

AlAmantea, 2025-02-07:
> Update the firmware to the current version of klipper

Is there an easy way to do this? I'm not familiar with the process of updating as this is my first Klipper printer.

linouxis9, 2025-04-02:
Any updates on the matter please? @QIDITECH