QIDITECH/QIDI_PLUS4 #29: Filament not able to load into extruder.

Channel-VR, 2024-11-15:
When trying to load a new filament or even unload filament the extruder bunches  back and forward and is not able to load or unload filament.
Check this video  : [https://youtube.com/shorts/6AW4GeOc9Ms?feature=share](url)

I unscrewed the extruder but there is nothing clogged and the wheels are all freely turning.

Any ideas ?  Thanks.

67RS427, 2024-11-15:
You probably have a clog... Heat that sucker up to 300˚C and see if it will push out. Otherwise follow the instructions in the Wiki and use a needle or drill bit.

Channel-VR, 2024-11-15:
Nope the entire extruder is empty . Problem is that the filament isn't even reaching the heat element.
The wheels that supposed to grab the filament don't turn they just go back and forth a few mm .

QIDITECH, 2024-11-16:
The extruder gear is not rotating, please check whether the gear and motor mesh is normal, if there is damage, please contact after-sales service for replacement.

closed 2024-11-20 by QIDITECH