QIDITECH/QIDI_PLUS4 #37: Plus4_V1.6.0 Version Release broke the octoeverywhere connection in ORCASlicer

nilsolovp, 2024-11-18:
Orca slicer can`t use my octoeverywhere connection after the update to Plus4_V1.6.0 Version Release.
I have no problem connecting and uploading a file thru the connection but the fluid interface function (devices) is broken.

nilsolovp, 2024-11-18:
It seems to be the fluid interface all together that doesn't work. Not simply a octoeverywhere issue.

QIDITECH, 2024-11-19:
Please try reinstall octoeverywhere.

nilsolovp, 2024-11-19:
Well I found out that I only needed a freshly prepare usb drive with the update for the printer to accept it and do a new try of installing the firmware. 
Unfortunately I still have no access to fluid. 
Not over local network, not thru octoeverywhere.

QIDITECH, 2024-11-20:
From the current situation, the most likely will come from octoeverywhere. We will try to reproduce the problem.
And what happens if you connect to fluidd over a LAN? Please take a screenshot and tell me.

nilsolovp, 2024-11-20:
Over lan I can reach Fluid login page and enter my credentials. Then I reach a blank page with fluid logo and drop down menu. However there is nothing there. 

Same issue with quidi link. I reach a blank page. 

Mobileraker can however control everything that’s built in this app. And this connection is over octoeverywhere.

QIDITECH, 2024-11-21:
It seems klipper and moonraker are working fine, only fluidd is abnormal.
Contact after-sales service to obtain the system image file and installation method.

dssamusaran, 2024-11-24:
I have the exact same issue. Fluidd is broken since update. 

I contacted support and I'm waiting for an answer as well.

The reinstall method and files should be somewhere on the wiki or github for easier access for people.

AlAmantea, 2024-11-28:
Same problem here.
Even Qidi Studio won't show the fluidd interface.
The connection does test okay, and it says "Connection to Moonraker okay", but when trying to display the fluid UI it goes to a blank page and just sits there.
Qidi Link also will not connect to the printer
Link or LAN mode does not make any difference.
It's like the moonraker server is running, but the UI is crashed in the printer.

dssamusaran, 2024-11-28:
I had some news today from support. They basicaly do not know what happened with some updates and they're going to send a new emmc to replace the old one.

Quite a drastic mesure but I guess it'll work and get the machine back up and running.

AlAmantea, 2024-11-28:
@dssamusaran Sounds like I need to contact support myself then

dssamusaran, 2024-11-28:
@AlAmantea I think so to have this solved asap.
Be aware that you might encouter custom duty fees to get the part, but keep your receipt and send it back to support, they'll refund that no question asked. ;)

Maybe at some point they'll find a way to reinstall without changing the emmc altogether but I would contact them right now if I were you. :)

It takes a few days but they're keen to help.

AlAmantea, 2024-11-28:
@dssamusaran Thanks. I'll send them an email right now.

QIDITECH, 2024-11-29:
@AlAmantea We are very sorry, we have repeatedly tested on multiple devices, but the problem still cannot be repeated, please refer to the link below to try to update emmc, or contact after-sales service.
https://wiki.qidi3d.com/en/Memo/flash-emmc

Makerdust, 2024-11-29:
The problem may be the Wi-Fi adapter. I had similar connectivity issues with my pc when its adapter was failing, so I tried swapping the printer's stock dongle for one I had laying around and the problem immediately went away.

nilsolovp, 2024-11-29:
I’ve tried replacing my stock one with TP-Link AC600.
It doesn’t fix our specific issue. 
And starting the printer without any WiFi dongle doesn’t fix it unfortunately.

dssamusaran, 2024-11-29:
@QIDITECH Is this adapter suitable to do the flashing?
[image]

I don't see clearly the kind of emmc you're using in the documentation. And I don't find an micro-sd adapter that looks compatible.

QIDITECH, 2024-11-30:
@dssamusaran Yes, this one works too.

AlAmantea, 2024-12-01:
@QIDITECH I have ordered a emmc adapter, also spoke with after-sales support.
I believe they are sending me a new emmc to replace the one I currently have installed to try and resolve the problem.
If you are anyone else on your dev team is willing to remotely login to diagnose and resolve this issue, I would certainly be willing to do that in order to more quickly fix the problem that a lot of us are having.
Please contact me through email, or have after sales support or marketing contact me directly and we can set up something to do this.

QIDITECH, 2024-12-02:
Thank you for your cooperation, although we were unable to reproduce the issue on X-PLUS4, we did experience the issue on X-MAX3. Although the probability of occurrence of this problem is low and cannot be stably repeated, two effective solutions are known. 1. Replace the emmc and update the firmware. 2. Use SSH to connect to the printer and replace the fluidd folder.
We will post the details on the wiki when we have finished.

AlAmantea, 2024-12-06:
@QIDITECH Replacing the fluidd folder did not make any difference at all. I believe after-sales support is sending me a replacement emmc.
I will post again after it is replaced.

nilsolovp, 2024-12-06:
As AlAmantea said, I also replaced the fluidd folder in a try to fix it and it didn’t fix our issue. I’m replacing the emmc.

AlAmantea, 2024-12-06:
Finished replacing the eMMC. Issue is resolved for me with that replacement.

QIDITECH, 2024-12-07:
Please let me know if you will replace fluidd folder directly or delete and import fluidd folder first?
Are fluidd folders named with uppercase letters?

AlAmantea, 2024-12-07:
@QIDITECH The one on the printer was lowercase, and the one on the github was "Fluidd" with an uppercase F.
I renamed the original fluidd folder to fluidd.old, then uploaded the new fluidd folder, and changing the name to all lowercase to match the original.
When that did not have any success, I tried with an uppercase F as shown in the github folder, but had no joy.

While my issue is since resolved by replacing the eMMC, I am still curious as to what caused the issue to begin with.
I do have an eMMC reader, and the capability to copy files and folders from the original eMMC if there are any logs or files you want to look at.

nilsolovp, 2024-12-07:
Ahh. These thing might matter.

I first renamed the old one and added the folder ”Fluidd” from GitHub then restarted the printer. That didn’t work. And it even made it worse.
Then I realized that the original folder was spelled with lower case ”fluidd” in the printer and I renamed the folder. 
This brought me back to the original state were it seems to work but never really loads the fluidd interface.
After that I tried deleting the fluidd folder and transferring the folder already renamed.
No difference.

In the mail I got from the support was an pdf with the example on how the procedure was to replace the klipper folder, and moonraker. By deleting these folder and replacing them.
So I followed this PDF and this unfortunately resulted in that now I can’t start the printer. It simply reaches a point where it showed an error on the display.  ”The system started abnormally! Please turn of the machine and wait for a while before restarting”

So I’m at the point where I need to replace the emmc to get a working printer at all.

AlAmantea, 2024-12-07:
@nilsolovp That really sucks.
One thing to note is that when I did replace the eMMC, at first boot, the system still showed the latest firmware update being installed, so I did not have to go through any firmware updates. In fact, it didn't even give me an option to do so.

The eMMC replacement is very easy to do. 6 screws to remove the back cover, and then 2 screws to remove on the eMMC. pop it out, install the new one, and reinstall the screws. It took me less than 5 minutes.

I do plan on remaining engaged for this issue, since I have full access to both the new eMMC through SSH, and the old one through an eMMC reader.

Also of note, before I relaced the eMMC, the printer would still boot normally, and I was still able to send prints, even after breaking the fluidd folder and as you did, the moonraker folder as well, by attempting replacement. I just did not have any UI connection for the device through the slicer, app, or browser.

QIDITECH, 2024-12-09:
@AlAmantea The emmc we sent has been refreshed with the latest firmware, so you do not need to refresh it again.
Thank you for your cooperation. Since there is no error in the fluidd file, please confirm whether the proxy file is damaged.
[image]
[image]

AlAmantea, 2024-12-09:
@QIDITECH There does not appear to be any damage to the files or locations shown in your images. 
Both folders are there, and contain the files listed. I am also able to read and display all 3 files normally.

nilsolovp, 2024-12-11:
The emmc that was sent to me by the support fixed the issue.
I replaced it tonight.

Gybk, 2025-01-29:
I also have this problem
wiki and fluidd remove didn't help me

Gybk, 2025-01-29:
[image]

phrac, 2025-02-18:
It looks like the way to fix it is to downgrade the version of Fluidd to an older version. At some point, Fluidd stopped working with the version of Moonraker that is supplied with the printer.

6902peabody, 2025-03-03:
I just got a new Plus 4, I did a firmware upgrade to 1.6 and experienced the exact same issues stated in this thread - I purchased my own emmc, flashed it with the qidi provided image from their wiki page, upon booting my printer said the SOC and UI firmware didn't match and that I need to manually update to 1.6 firmware - well guess what - it broke, just like my original one did.  

What kills me here is that qidi tech says they cant reproduce the issue but I did, twice and its exactly like what the OP experienced.

Manually upgrade the firmware by downloading the 1.6 file to a USB, run the upgrade, let your printer run for about 24 hours and you will lose connectivity to fluidd.  Its not immediate because after I did the upgrade it worked fine, but the next morning I couldn't connect again.

Qidi needs to upload an image of the emmc that they are shipping to people that has the full 1.6 firmware upgrade already installed.  The .img file they provide to image the emmc is not updated, and forces you to upgrade, thus breaking fluidd.  I too am now waiting for a new emmc from qidi to be shipped. I will immediately image that emmc so I have a good working image to burn in the future.

aryeh26, 2025-03-25:
same with me! how is this still not fixed!

6902peabody, 2025-03-25:
> Same issue here with my new Plus 4. Worked perfectly for a few days. Now device page in Qidi studio (1.10.2.76) windows says connection is OK but a white page where the details (printer controls and camera image) were normally displayed.
> 
> The printer itself still works fine, Qidi studio sends the file, reports printing status and % complete, but that is all I cannot believe a replacement part is needed to fix this... surely?

Technically you don't need a replacement part, you could just pull your current emmc out and reflash it - but for piece of mind I purchased a second emmc to test the process out.  I just posted how I fixed the issue with my printer and can tell you it has been working without issue since I did it.  You can always send an email to qidi but its kind of a pain, they don't get right back with you - its always a few days to respond, then they have questions and you reply right away and its a few more days before they respond - in the end they did send me a replacement emmc that should have fixed the problem, but I was able to purchase one from amazon and figure out the fix before they sent me the replacement.  Amazon is way quicker than waiting for qidi.  I really like this printer - once its dialed in its really good.  About ready to do the beacon update on mine.  glad I have the extra emmc's now in case that goes sideways.

phrac, 2025-03-25:
A quick and easy way to fix this is to just downgrade your version of fluidd:

SSH into your printer and perform the following:
```
cd ~
mv fluidd fluidd.old
mkdir fluidd
cd fluidd
wget https://github.com/fluidd-core/fluidd/releases/download/v1.30.1/fluidd.zip
unzip fluidd.zip
```

6902peabody, 2025-03-26:
> It looks like the way to fix it is to downgrade the version of Fluidd to an older version. At some point, Fluidd stopped working with the version of Moonraker that is supplied with the printer.

version of fluidd on qidi firmware 1.6 is  v1.30.4-221de1fe, and its not broken it works fine. There is an issue when updating both the SOC and UI, especially if the SOC is already at 1.6 - the updater isn't smart enough to check to see if the update is at the same version and just not do anything, and whatever it does screws up fluidd.  I tried rolling back fluidd while on 1.6 firmware to the version of fluidd from 1.4 firmware and it worked for a while, but basically when the sessions got logged out (web or via qidi studio) you were right back into it not allowing you to get in.

QIDITECH, 2025-03-28:
I am sorry that it has taken so long for us to rediscover the problem and find a solution.
The problem is caused by an error in the fluidd database, which can be resolved by deleting the database where fluidd communicates with moonraker and restarting the printer. Please refer to the last section of the link below for details
https://wiki.qidi3d.com/en/software/qidi-studio/troubleshooting/blank-page

JLRGURU, 2025-03-28:
Great ro now! I just reflashed two time the emmc for this problem in the past!

Thank you!