QIDITECH/QIDI_PLUS4 #39: V1.60 LAN Connection only setting turns off the LAN rather than turning it ON.

atrozgrifo, 2024-11-20:
I upgraded from 1.4.2 to 1.60 and my LAN connection no longer worked (I don't use WiFi).  After some experimentation I found out that if I turned OFF the LAN Connection Only (I had it on for the last few releases) setting in the Network panel through the LCD the Plus 4 would now communicate on the network.

QIDITECH, 2024-11-21:
The lan connection only button means turning off Internet functionality, not switching between wifi and wired connections. Then turn on the feature in the version, there are still some communication threads that are not closed, and now the new version will fix it.

atrozgrifo, 2024-11-21:
Ah, OK.   I thought that was what it was for when I first setup the printer but it didn't work that way.  My printer and computer are on different subnets and it wasn't blocked then so I figured it was mislabeled.

closed 2024-11-21 by atrozgrifo