QIDITECH/QIDI_PLUS4 #40: Incorrect initial Z offset with ASA and ABS on the first power-on.

budfox22, 2024-11-22:
Hi All,

I would like to report an issue I am experiencing with printing materials like ASA, specifically related to an incorrect Z offset.

When I power on the printer, after the automatic nozzle cleaning and bed leveling, the first print always has the Z offset too close to the bed, requiring me to restart the print. 

On the second attempt, everything works perfectly. 

I’d like to point out that this issue only occurs after the first power-on or reboot of the printer, particularly with higher-temperature filaments like ASA, ABS, etc.

Example:

Qidi Studio and Qidi ASA filament

1. Power on the printer.

2. Heat the bed to 90°C for 15 minutes.

3. Send an STL file from Qidi Studio to the printer.

4. The printer performs nozzle cleaning and bed leveling.

5. Start the print, and the Z offset is too close to the bed.

6. Stop the print.

7. Send a new print, and everything works fine.

Thanks

Drake0815, 2024-11-22:
sounds similar to the problem I am currently having with printing ABS

[image]
[image]

Ken6432, 2024-11-22:
Have you tried this process from the Plus4 Community GitHub?  I'm not sure if it is sanctioned by Qidi, but it might help you out for now.

https://github.com/qidi-community/Plus4-Wiki/blob/main/content%2Fmaking-z-offset-permanent%2FREADME.md

budfox22, 2024-11-22:
Thank you for your advice.

No, I haven’t tried it yet because I preferred to wait for an official response from Qidi.

I don’t understand why this issue only occurs on the first power-up or restart of the printer when using filaments like ASA, ABS, etc.

From the second print onward, the Z-offset is perfectly fine.

At the moment, the only solution I have is to stop the print and restart it or adjust the Z-offset live.

sonny486, 2024-11-22:
Yes I have the same issue, I have my Z-offset set to + .05.  That seems to be good for me.

Not sure why it's not working well by default.

QIDITECH, 2024-11-23:
The high temperature generated by the chamber heater may affect the sensitivity of the piezoelectric sheet, some positively and some negatively. We adjusted the starting time of the chamber heater, but this did not solve the problem.
We are improving the process of piezoelectric plates and have increased their yield.
You can contact after-sales service to get an update on this progress so that you can replace it as soon as possible.
Or you can adjust the gcode macro to resume starting the chamber heater before leveling.

budfox22, 2024-11-23:
Thank you for the detailed explanation.

closed 2024-11-25 by QIDITECH

budfox22, 2024-11-25:
I noticed that you have modified the comment you previously made with a substantial change.

Initially, you wrote:

"....We are improving the process of piezoelectric plates and have increased their yield.
You can contact after-sales service to get an update on this progress so that you can replace it as soon as possible...."

Then it was changed to:

".... We are improving the process of piezoelectric plates and have increased their yield.
We recommend you provide a detailed description of the first-layer printing issue you are experiencing. Including photos or videos would help us better understand the problem and provide targeted support...."

So, what should I do now? Should I contact after-sales support, or should I write here?

As already mentioned, the main issue is that I need to adjust the Z-offset every time I restart or turn on the printer.
Even after performing all the necessary calibrations, it seems the Z-offset varies each time.

This happens with filaments like ABS, ASA, etc., using a bed temperature of 100°C and a nozzle temperature of 255°C.

Firmware version: 1.6.0

Thank you.

QIDITECH, 2024-11-26:
We have fully understood the problem, please contact the after-sales service to help you solve the follow-up problems.

Drake0815, 2024-11-26:
> em, please contact the after-sales service to help you solve the follow-up problems.

This is what i do from my side...and they telling me i need to set the zoffset for each material manually!?
This is not acceptable i think.

QIDITECH, 2024-11-27:
If you use a different email address than your github username, we won't be able to determine which one is you.
Did you describe the problem clearly?
We will check the mailbox to confirm again.

bogdanm, 2024-12-11:
@QIDITECH Please consider leaving this issue open to track its progress. It's not relevant only to the reporter, but also to people who already own the machine or are looking to get one (such as myself). Keeping it open would allow eveyone interested to track its progress. Thank you.

Drake0815, 2024-12-11:
> I noticed that you have modified the comment you previously made with a substantial change.
>
>
>
> Initially, you wrote:
>
>
>
> "....We are improving the process of piezoelectric plates and have increased [...]

I get from Qidi Support new Heatbed Level Sensors they I have installed today. After this I have reinstall also the new Formware on the printer.
Now I'm realy happy
I have had today my best first layer with ABS, and without Offset change.

[image]

bogdanm, 2024-12-11:
@Drake0815 Glad the issue is fixed for you! That looks really nice.

budfox22, 2024-12-12:
I don’t.

 Even after replacing the sensors, I still have the same issue with the Z-offset when printing with ABS and ASA.

For me, nothing has changed.

bogdanm, 2024-12-12:
@budfox22 If it's not too much trouble, please keep us (me ? :) ) updated with your progress. I won't purchase this printer until there's some kind of resolution on this particular issue, since printing ABS/ASA comfortably is one of the main reasons for purchase. Thank you.

budfox22, 2024-12-12:
@bogdanm 

Sure!

I can tell you that the Plus 4 prints ABS and ASA perfectly. I just have this annoying issue where I have to adjust the Z-offset during the first layer every time.

SupaCoder, 2025-01-19:
@budfox22 

Sorry but this:

> I have to adjust the Z-offset during the first layer every time.

is not acceptable on a printer like this, if you are not switching filaments (and even if you are) the printer should perform consistently EVERY TIME. Im currently having issues with my Plus 4 as well, im running v1.1.2 firmware and have come here to see if v1.6.0 would solve things but it seems like theres the same issues here. Im also having to live adjust z_offset every time i print with ABS, and that is very disapointing for a printer of this price.

nemith, 2025-01-22:
Brand new QIDI Plus 4 and printed with ASA and yes, my Z-offset was way too low.