QIDITECH/QIDI_PLUS4 #42: Fluidd interface not working since 1.6.0 update

dssamusaran, 2024-11-23:
Since doing the online update from the machine from 1.4.3 to 1.6.0 the fluidd interface is not working anymore.

Ping is answering, and SSH as well, so the network is OK.

But when trying to load the fluidd web page only a blank white page show up.
Tested from multiple PC/Phone devices.
With Chrome/Edge/Firefox.
QidiStudio in the device tab is showing the same blank white page.

Machine was unmodified and fully stock.

->Multiple reboot done (full power off/on) does not solve the issue.
-> Factory reset from the screen does not solve the issue.
-> The recovery from this documentation does not solve the issues. => https://wiki.qidi3d.com/en/PLUS4/Manual/firmware-update

Ken6432, 2024-11-23:
It may not be related, but there is a known issue (I think it's #32) related to the wifi dongle being in place when using Ethernet.  If this is your situation, you might try removing the wifi dongle (you will need to remove the small back panel that holds the cooling fan).

As I say, it may not be related, but I wanted to pass this along.

nilsolovp, 2024-11-24:
This issue is similar to mine (issue 37), and removing the WiFi dongle doesn’t help unfortunately.

dssamusaran, 2024-11-24:
Hello, I tried removing the dongle and indeed issue is still there.

I read your issue and yes it seems to be the same.
Will close my issue as it's a duplicate of Issue #37

Having to go through support to get the firmware and install method seems counterproductive. Will be easier to get it on the wiki or github.

dssamusaran, 2024-11-24:
Duplicate of #37

dssamusaran, 2024-11-24:
Duplicate of #37

closed 2024-11-24 by dssamusaran