QIDITECH/QIDI_PLUS4 #45: chamber heating macro for 1.4.3 (wiki) still needed/useful in 1.6.0?

mrambossek, 2024-11-29:
in the previous release (1.4.3) i used [these steps](https://github.com/qidi-community/Plus4-Wiki/tree/main/content/tuning-for-40-percent-heater-power) to "tune" the chamber heating process.

it seems like 1.6.0 changed the PRINT_START macro slightly? not sure, i didnt keep a backup from before 1.4.3 unfortunately.

do the steps in the wiki still make sense? if it is a good idea, why is the firmware not coming with this as default now?

QIDITECH, 2024-11-30:
This is not an official document, so please think carefully before following it.

mrambossek, 2024-11-30:
> This is not an official document, so please think carefully before following it.

understood. so the "official" way is still what the firmware does anyway.. and the community post is just "an alternative way" that might or might not work. which apparently is not "just better", or at least not validated, otherwise it would have been used by the official firmware....

thanks for clarification :)

closed 2024-11-30 by mrambossek