QIDITECH/QIDI_PLUS4 #46: M600 for manual filament change - using M603?

mrambossek, 2024-11-30:
this is my first kipper printer and thus my understanding of how it all works and connects is still rather superficial.

the official "Box" for automatic multicolor printing is not yet available; but for some simple models, especially something like "writing" on the surface, manually changing a few times is often reasonable.

i tried using M0 in the "Change filament G-Code" field in orca slicer, which works fine, but does not cut the filament. in QIDISlicer, there are boxes for "Tool change G-Code" (empty) and "Color Change G-Code" (just M0), which probably means no cutting happens either (did not try).

would it be possible to implement an M600 macro in the firmware in the future, that basically does M603 + M0? i dont fully understand all the steps the current M603 macro does. would be great if we could have this "out of the box" so that we can just put M600 into the "Change filament G-Code" box and it works :)

QIDITECH, 2024-12-02:
Yes, adding the M603 before the M0 can automatically cut off supplies, but this may not work perfectly.
You can refer to M603 and write only the necessary instructions.
[image]

mrambossek, 2024-12-02:
thanks for the input. i am currently using this:

```
[gcode_macro M600]
description: Change Filament (M603 + M0)
gcode:
    M118 Change Filament (cut, then pause)
    {% set accel = printer.toolhead.max_accel|int %}
    _CG28
    M204 S10000
    G1 Y150 F9000
    G1 X305 F9000
    G1 Y20 F9000
    G1 Y5 F3000
    G4 P500
    G1 Y20 F9000
    G1 Y5 F3000
    G4 P500
    G1 Y20 F9000
    M204 S{accel}

    {% if printer.extruder.can_extrude %}
    PAUSE
    {% endif %}
```

and it works, but it is not as "nice and seamless" as swapping filaments on other printers. definitely looking forward for the BOX... :)