QIDITECH/QIDI_PLUS4 #48: Xindi consumes excessive CPU resources

stew675, 2024-12-25:
Xindi appears to do something akin to CPU busy-waiting for events.  On my printers,  both 3-series and on the Plus 4, it is consuming almost an entire CPU core by itself.

As a programmer, can I suggest that xindi be tweaked to be event driven with the use of the `epoll()` system call such that it is only responding to events rather than busy polling for activity?  This should be a relatively minor change to implement and would greatly reduce the system load on the CPU and memory bus, and should give the CPU better capability to handle higher print speeds.

If Qidi would like assistance with implementing this, I can offer my services to modify xindi to make this a reality.

Unkleskunky, 2024-12-25:
Hello,

Is this why the plus 4 will not switch tabs or do anything it seems until
it has completed the prior task? For example plus 4 will not switch to the
USB tab in "doc" until thumbnails load? Or z position changes until prior
movement or process is complete?

Thanks

C

QIDITECH, 2024-12-26:
Waiting for events is a must, otherwise, frequent operations can lead to errors and eventually clutter the UI.

closed 2024-12-26 by QIDITECH

stew675, 2024-12-31:
> Waiting for events is a must, otherwise, frequent operations can lead to errors and eventually clutter the UI.

I believe that there has been a misunderstanding.  xindi will still wait for events, but it releases the CPU until the OS wakes the xindi process up when an event occurs.  This is achieved by using the epoll() system call.  It prevents the need to do the extremely wasteful busy waiting cycle that xindi is presently employing