QIDITECH/QIDI_PLUS4 #49: Re-opening: Xindi consumes excessive CPU resources

stew675, 2024-12-31:
Re-opening as there appears to be a misunderstanding.

I believe that there has been a misunderstanding. xindi will still wait for events, but it releases the CPU until the OS wakes the xindi process up when an event occurs. This is achieved by using the epoll() system call. It prevents the need to do the extremely wasteful busy waiting cycle that xindi is presently employing

Xindi appears to do something akin to CPU busy-waiting for events. On my printers, both 3-series and on the Plus 4, it is consuming almost an entire CPU core by itself.

As a programmer, can I suggest that xindi be tweaked to be event driven with the use of the epoll() system call such that it is only responding to events rather than busy polling for activity? This should be a relatively minor change to implement and would greatly reduce the system load on the CPU and memory bus, and should give the CPU better capability to handle higher print speeds.

If Qidi would like assistance with implementing this, I can offer my services to modify xindi to make this a reality.

To be very clear, with this change, xindi will STILL wait for events just like before, but it will be done in a FAR more efficient manner.

Please read the following for an example of this programming method: http://web.stanford.edu/~hhli/CS110Notes/CS110NotesCollection/Topic%204%20Networking%20(5).html

Doing this will make xindi use perhaps 95% less CPU than it presently is.

stew675, 2025-01-04:
> Stew, doesn't doing the " mks@mkspi:~$ pidof xindi" to get value and then using "mks@mkspi:~$ sudo renice 10 -p 1997" to lower value to 10 would do the same ?

It does help, but it doesn't change that the `xindi` process is busy waiting.

Over the last two days I've actually developed a better way to minimise what `xindi` is doing, as well as improve the CPU processing performance of the Rockchip CPU on the Plus4 overall.

It's all documented here: https://github.com/qidi-community/Plus4-Wiki/blob/main/content/system-tuning/README.md

This puts the CPU into performance scheduling mode, forces a fixed 1200Mhz CPU frequency, and isolates `xindi`, `mjped_streamer` and `nginx` to CPU core 3 and nices them down a little, and also isolates Klipper to CPU cores 0, 1 and 2.  This completely segregates the poorly behaving xindi.

It actually improves the performance of the video streaming as well.  I would presume due to better CPU cache coherency.  It's now possible to run 1920x1080p video capture at 25fps, and this doesn't affect Klipper ability to get a hold of the CPU any more.

I'm also sort of toying with if Moonraker should also be isolated to CPU Core 3, and this would basically allow Klipper near uncontested access to 3 of the 4 CPU cores.

Again, with these changes, overall responsiveness of literally everything has improved.

Unkleskunky, 2025-01-04:
> Stew, doesn't doing the " ***@***.***:$ pidof xindi" to get value and then
> using ***@***.***:$ sudo renice 10 -p 1997" to lower value to 10 would do

Sounds awesome,

Thanks for your efforts!

US

stew675, 2025-01-08:
> Over the last two days I've actually developed a better way to minimise what `xindi` is doing, as well as improve the CPU processing performance of the Rockchip CPU on the Plus4 overall.
>
> It's all [...]

To any of you who were using my script, I tracked down and fixed an issue with regards to `xindi` and the Printer's Display.

Please re-download the script to pick up the bug fix changes.

Unkleskunky, 2025-01-09:
Hi,

I am interested in what you are doing here. Do you have a walkthrough I can
follow? This is way out of my toolshed and I don't want to brick my Plus 4.

Thanks

C

mrambossek, 2025-01-09:
his post pretty much describes in detail a) [what to do -> 3](https://github.com/qidi-community/Plus4-Wiki/blob/main/content/system-tuning/README.md#installing-the-tuning-script) and b) what it does (below that)

i mean theoretically it is of course possible to break stuff if you paste stuff wrong etc. imo, if you are not familiar enough with linux for these commands, i woulnd't touch it for now - the "gain" is not *that* great :)

you can always recover if you have an eMMC flash adapter *or* a second, already written-to, eMMC module from QIDI or wherever.

mrambossek, 2025-01-09:
that being said, i am using the script myself for a few days now and did not notice any downsides. i did not have a lot of free time to spend with the printer in the first place, so i did not see any "noticeable" improvements either. but as far as i can judge, its done in a very safe way that should not "brick" anything.

Ken6432, 2025-01-09:
I have been using the print a bit since adding the changes and have not seen any issues.  I just reinstalled the update this morning, so I haven't run any since.  I agree that the likelihood of bricking the machine is incredibly low.  It is a pretty straightforward approach.

Unkleskunky, 2025-01-09:
Cool, I may give it a go.

No noticeable improvement to the video?

C

Ken6432, 2025-01-09:
I saw no difference in the video.  I increased my resolution today and am still getting high frame rates when watching through my iPad browser

Unkleskunky, 2025-01-10:
interesting,

Are you hardwired at the printer?
I'm getting like 4 fps over wifi. Its fine but better would be nice.

C

Ken6432, 2025-01-10:
Yeah, I haven't figured mine out really.  My printer is hardwired.  On my computer (also hardwired and plugged in the same router) I get 4 fps all the time.  This is consistent between connecting through different browsers or the slicer.  However, if I connect through the browser from my iPad (obviously wirelessly) I get 15 fps consistently.  I have no clue why that is

Unkleskunky, 2025-01-10:
Nice!