QIDITECH/QIDI_PLUS4 #51: Qidi Studio Login/Registration doesn´t work

Alphat13r, 2025-01-10:
Since last week in Qidi Studio when I click on "Tool" I can see my Qidi X-Plus 4 and also connect and see the status.
But there is no Dashboard/Fluidd visible. Maybe because I need to Login, but this doesn´t work. There is always just a White Screen.
[image]

Alphat13r, 2025-01-10:
Additional Information: Login via Web or Mobile is possible...Qidi Studio was already uninstalled, but still doesn´t work :(

QIDITECH, 2025-01-11:
No login is required for local connections. Please confirm:
1. Check whether fluidd can be opened through the browser
2. Check whether the WebView2Loader.dll file in the installation directory exists

Alphat13r, 2025-01-11:
Hello there,thanks for your answer. Fluid could be opened in web Browser and also on mobile device App.Webview2Loader.dll also exists at the Installation path.I also tried to uninstall, get the newest Version and Install it. But still there is No Login possible (White Window opened) and also No printer Kamera or dashboard Displayed.--Gesendet mit der GMX Mail App

QIDITECH, 2025-01-13:
It seems that there is a problem with the cache of the software, please try to delete the device and re-enter the IP, if it still does not work, please go to the following path, and delete the cache file EBWebView after closing the software. The AppDate folder is usually a hidden folder, so turn on the View Hidden files option.
[image]

Alphat13r, 2025-01-13:
Hello there,Problem solved!I tried all with the Qidi Studio Software. Also delete and add the printer.Nothing helps. Then I tried the Orca slicer and within installing this Software IT also installed another App called webview... Or Something else.Seems that there was a Change in my Windows and the Qidi Studio doesn't Check this addon, which the Setup from Orca checked.Maybe you should implement this to your Installation File ;)Now the Qidi Studio also works fine :)--Gesendet mit der GMX Mail App

image.png (view on web)

QIDITECH, 2025-01-14:
Thanks for your feedback, we will add the installation of windows plug-in in the next version of the installation package.

closed 2025-01-14 by QIDITECH