QIDITECH/QIDI_PLUS4 #52: Timelapse with park position 'Top Left' crashes the printhead in the corner at first shot.

sunbowch, 2025-01-16:
Most is on the title.
Purge line is printed before first image is taken.
Then the head moves into the right direction, but crashes in the corner as if the limits were not defined.
I stopped the experiment here... I wasn't able to react before the printer starts the first layer, but it was drifted about 50 mm apart on both axes.
I don't want to try other positions unless you send me some spare belts and other parts likely to be damaged by this issue.
Tested manually defined park position works without crashing.

QIDITECH, 2025-01-17:
By default, when time-lapse photography, the nozzle will move to the position of the mouth, obviously you edited the code to move, then you need to note that there is a small area in the upper left corner for the non-printable area.

sunbowch, 2025-01-17:
Hi!Sorry, but I did not edit the code... I just choose the option 'Top left' in the settings-timelapse window of fluidd software...But after having looked at timelapse.cfg macro code it seems that the park position doesn't take the 'non printables areas' into account during the calculation of min/max park.x/park.y position...Now there is 2 options:1) Change the code and take into account the non-printable array in the calculation of 'top left' and 'front right' positions... Others should be fine. Or to make it simpler add a suitable and fixed offset in the calculation of the park position so that the head stays within a rectangular area outside of the non-printable areas.2) Suppress the 'top left, top right, front left, front right' options and leave only the 'custom position' option...I can code one or the other and send a pull request... But I think that this effort is worth a set of new belts... Mine have sustained an heavy overload while I was discovering the bug... And this has undoubtedly affected their lifespan.Sincerely

QIDITECH, 2025-01-18:
OK, thank you for your feedback, we will optimize this issue in the next version. As for the belt, please contact after-sales service for replacement.

sunbowch, 2025-01-26:
It seems that the 'timelapse' option offered when printing a part through the touchscreen has some unresolved interference with klipper / timelapse.cfg.
One absolute rule of using open source code is to never ever break its basic functions... So the printer's gui 'timelapse' option should simply follow the rules and settings defined through fluidd/settings/timelapse... And the timelapse.cfg macro defining 'top left, top right, front left, front right'  position must take into account the non-printable areas...
And don't tell me that somewhere in a manual it is written not to use klipper/fluidd 'timelapse' settings when using timelapse option directly...