QIDITECH/QIDI_PLUS4 #56: Chamber Heating Before/During Kamp Probing

81gtun4, 2025-01-25:
In the earlier firmware releases, (eg 1.4), the chamber and the bed would heat up together before the printer executed the Kamp bed probing. I understand that this was changed as a safety measure for those who were on the faulty SSR. However, after moving to the new SSR, there is not a concern here. Also, it is giving people first layer issues (myself included).

I have a plus 4 on firmware 1.4 (with a new SSR) and it preheats the chamber and the bed. The chamber power setting is at .7 and the thermal protection sensor maxes out at 110c during start up and then maintains 55-63 degrees throughout the print. This is for a print of 55 Celsius chamber and 104 Celsius heat bed.

My other plus 4 that is giving me issues is on firmware 1.6 and also has the updated SSR. It gives me all sorts of first layer issues even thought the bed level is near perfection at below .1 variance. The printer on 1.6 firmware thermal protection sensor runs about 5 degrees Celsius higher.

Can you please add the option for chamber heating prior to and during the kamp for improved first layers and adhesion for those who have updated the SSR.  The new SSR does not get above 30-40 C in my testing.

81gtun4, 2025-01-25:
Otherwise, can you send me a copy of 1.4 firmware so I can run my other qidi machines on it. My 1.4 firmware machine is perfection for the PC prints I have to do.

QIDITECH, 2025-01-27:
You can do this by modifying the macro PRINT_START.
[image]

81gtun4, 2025-01-27:
Thank you for your reply.

I appreciate your provided direction. Uncommenting that line does make the machine pause and wait for the chamber to get up to temperature before beginning probing, however, once probing begins the chamber shuts off and the temperature start dropping reasonably fast. It drops over 10c during the probing.

Do you have the gcode solution for keeping the bed on during the probing as well?

There would be value in having this option as the default as I've seen a fair number of others posting online with this issue. 

I've also read that higher chamber temperatures (60c+) can possibly interfere with the accuracy of the probing. If this is the case that the higher probing temps are of concern, then it would be great to see an option for a probing temp that is at least higher than the ambient ~30c as the default.

The 1.4 firmware also has this commented out as per your screen shot, however, the behavior sets both the chamber and the bed to heat at the same time. Are you able to provide guidance on what gcode can achieve this? I have tried copying over the 1.4 gcode over to the 1.6 firmware and that did not solve it.

Thank you.

81gtun4, 2025-01-27:
is it just setting line 223 from "m141 s0" and changing it to "m141 S{chambertemp}   ?

QIDITECH, 2025-01-30:
Yes, but we recommend that you wait until the hot bed is heated before you start heating the chamber.

closed 2025-01-30 by QIDITECH