QIDITECH/QIDI_PLUS4 #57: Extruder pressure retaining arm default

FxWoody, 2025-02-07:
We found a big issue with the extruder pressure retaining arm that starts to deflect on long term heat exposure.
Issue is that the spring starts to loose strength of putting constant pressure on the arm to maintain an equalized pressure on the drive gears of the extruder and ends up skipping or causing jam filament between the gears.
This part of the extruder should be made out of aluminium and also have a "slack adjuster" on the extruder housing for the spring itself .
Reference of the issue : 

[image]

[(https://www.facebook.com/share/p/17L5xFChYq/)(url)]

Unkleskunky, 2025-02-07:
Yikes,

I would like to see the resolution to this.

The facebook link is broken for me.

thanks for bringing this to our attention.

QIDITECH, 2025-02-07:
Please contact after-sales service for replacement.

FxWoody, 2025-02-07:
> Please contact after-sales service for replacement.

I do not need to replace it on my machine yet. We need you ,to open a RNC "Rework of non conformity" with your engineering department and redesign this part in a stronger material and add an adjuster in the housing for the spring tensioner.
This problem, if not fix properly, will probably happen on EVERY printer's extruder sold :(

Unkleskunky, 2025-02-25:
> I did it a long time ago for community.

@abychko

Is this the same component in Plus 4?

Thanks for your efforts either way,

US

abychko, 2025-02-25:
> [@abychko](https://github.com/abychko)
> 
> Is this the same component in Plus 4?
> 
> Thanks for your efforts either way,
> 
> US
> […](#)

I don't know if it's the same, but I think there is no reason to introduce slightly different part. I've just ordered my Plus4, so can't compare them right now.
anyway, I uploaded a STEP file for this part, so it can be adjusted in any CAD software

FxWoody, 2025-02-25:
Looks like it's the same extruder part in deed. 
Tks for the link and file 👍

Unkleskunky, 2025-02-25:
Cool,

Thanks again

FxWoody, 2025-03-29:
In case someone need to have quick fix 👍
[Cantilever pressure arm](https://www.printables.com/model/1240903-qidi-plus-4-extruder-cantilever-arm)

abychko, 2025-03-29:
> In case someone need to have quick fix 👍 [Cantilever pressure arm](https://www.printables.com/model/1240903-qidi-plus-4-extruder-cantilever-arm)

looks like more sturdy than mine, good idea to add some material

Unkleskunky, 2025-03-29:
Awesome innovation, peeps.
Anybody price an aluminum one off?

rogernolan, 2025-06-14:
I have just been told by @QIDITECH that this is not a defect. This part is supposed to be bent.

abychko, 2025-06-14:
> I have just been told by [@QIDITECH](https://github.com/QIDITECH) that this is not a defect. This part is supposed to be bent.

that's strange. bent part can't deliver required pressure, at least it's true for my Plus4. from my experience it's better to print the part from pure ABS to have maximum rigidity, but for sure I made a couple of parts from CNC'd aluminium

rogernolan, 2025-06-14:
Seems like that message was a mistake and I’m getting a new extruder. Did you commission the aluminium part yourself or is there a supplier?

abychko, 2025-06-14:
> Seems like that message was a mistake and I’m getting a new extruder. Did you commission the aluminium part yourself or is there a supplier?

I just ordered it locally using my stl/step design. some guy who can cnc aluminium did it for me
https://www.printables.com/model/902986-qidi-x-max3-x-plus3-x-plus4-extruder-part

rogernolan, 2025-07-08:
If anyone in the UK would like an aluminium version of this model, minimum orders mean I have a bunch of these I'm willing to sell for costs + postage.