QIDITECH/QIDI_PLUS4 #60: Z-Offset Calibration Fails at High Bed Temperatures (ABS Printing)

WoMec, 2025-02-17:
I would like to re-open issue #40 

**Description:**
The automatic Z-offset calibration on the Qidi Plus 4 fails when printing high-temperature materials such as ABS. The issue does not occur when printing low-temperature materials like PLA. The problem results in the nozzle being too low, dragging directly on the PEI sheet and causing damage.

**Steps to Reproduce:**
Heat the bed to 100°C or higher (e.g., for ABS printing).
Start a print with automatic Z-offset calibration enabled.
Observe that the nozzle is positioned too low, often dragging into the PEI sheet.
**Expected Behavior:**
The automatic Z-offset calibration should correctly compensate for the bed temperature and maintain a proper first-layer height.

**Actual Behavior:**
The nozzle moves too low after calibration.
First layer is extremely compressed or entirely scratched into the PEI sheet.
No issue occurs when printing at lower bed temperatures (e.g., PLA at 60°C).

**Console Logs & Evidence:**
Attached a full console log showing incorrect probe readings during Z-offset calibration.
Attached a photo of the first layer where the nozzle visibly scratches the PEI sheet.
[console_log.txt](https://github.com/user-attachments/files/18832911/console_log.txt)
[image]
[image]

**Additional Notes:**
Other users have reported similar issues: [Reddit thread](https://www.reddit.com/r/QidiTech3D/comments/1gr3o1n/qidi_plus_4_z_offset_issues/)
The printer is placed on a stable surface with no external vibrations.
A new PEI sheet has already been installed, but the issue persists.
It is suspected that the bed expansion due to higher temperatures might be affecting calibration.

**Environment:**
Printer Model: Qidi Plus 4
Firmware Version: 1.6.0
Slicer Used: QIDI Studio
Material: ABS
Bed Temperature: 90°C+

**Possible Solutions:**
Investigate how bed temperature affects Z-offset calibration.
Provide a firmware fix that compensates for thermal expansion.
Offer an option to disable auto Z-offset calibration after the first successful calibration.

Would appreciate any guidance or fixes for this issue.

Thanks!

nemith, 2025-02-18:
I had the same issue and got replacement load cells and now I have no problem.  

Two of the load cells weren't even connected and one the piezo was no longer connected to the load cell.    I think Qidi could put together something to test the load cells it would be great.

pravic, 2025-02-18:
I am curious whether the Platform Calibration or Auto Bed Leveling would help after preheating the bed to this high temperature. From the common sense - it should help, because the bed will be preheated and thus expanded due to the temperature.

guykuo, 2025-02-20:
Having bed adjustment screws too tight or too loose has also been implicated in rendering the piezo sensors inaccurate. This is in addition to this issue's calling out of thermal tolerance of the stock piezoelectric sensors. As bed temperature rises, the piezoelectric sensors also get hot and this reduces their signal output. Type of piezo material in sensor can limit operation of sensor to under 100C in low temp piezo material. This matches report of trouble with inaccurate z-touch probing at ASA print temperatures. Reduced piezo nozzle touch signal would causes late or non-existent sensing of nozzle touch. The resultant z-at-touch would be with bed too close to nozzle. That in turn will cause bed gouging during print with z-touch sensed too high.

A routine for testing z-touch piezo sensing accuracy at multiple bed temperatures would be invaluable.

nighteagle1974, 2025-02-22:
Hello, having same problem. New Printer bought on Januar 20th. Yesterday i print the benchy with the sample 
PLA-Filament and then i change the nozzle to the tungsten - carbide one and load ABS-GF. I want first calibrate Flow-Rate so i print the Flow-Rate object from Qidi Wiki with the Qidi Studio. The printer heat up chamber and bed and then begun bed-leveling.
Then start the print itself crashed the nozzle in the bed and destroy my PEI-Plate. Have contact Qidi-Support now and also have found this posts by reddit and now i'm here.

If the piezo-sensors are the problem because of the temperature - i wonder why Qidi has not found it before? Or is there also a bunch of printer outside without this problem? Then why the other have not the problem? I mean many people buy this one because of the printing Carbon or Glas Filaments - think they all need heated chamber?

I have demount the Piezo PCB-Boards on the bed and have resolder all components on it. Because i know they use lead-free solder. So 'm an engineer since 30 Years and know that this kind of solder is not for safety-electronics - because on heat they broke the connections to the components. There are many studies about this lead-free problem. So for space-technic and military they use lead-solder like all electronics in the past - that also the reason why today more electronic problems appear as in the past.

nighteagle1974, 2025-02-22:
So - i have found out - if i make manual heat chamber to 55 Degree and Bed to 90 Degree and wait 20minutes and then make Bed-Leveling over the Display from the Plus4 then it looks good. The don't bend the Pirnthead upside - what menas the bed goes to high.

If i then start a print over my usb - from Qidi-Studio it starts again the bed-LEveling and here it comes weird - it is pushing the Printhead upside - i can see the bending of the rails...

Is there another routine in the gcode of the file instead of the printer itself?

It looks like it is a software issue not the hardware-issue many posts talks about?

nighteagle1974, 2025-02-22:
Again it is weird - i had used Qidi Studio and now i'm up to Orca and print the same file for Flow-Rate Calibration - and it works. Bed-Levelling fine and print fine.

The difference i had seen was that with Qidi Studio the Chamber is heat before leveling. By Orca the Chamber heat up after leveling.
By the way - the camber shows 40 Degree from preheat... But in comparing to the first test with leveling over display - so it is only randomly fine leveling?

guykuo, 2025-02-22:
Nighteagle, I think you are seeing a difference between bed meshing and full sequence start of print. Start of print includes nozzle touch (bed moves up to touch nozzle) to find z-offset. Bed mesh does not necessarily need to nozzle touch because it is picking up relative height map, not absolute z at nozzle. 

I looked at the printer start g-code in PrusaSlicer, Orca, and QidiStudio. None of them command a bed mesh and probing in the printer startup g-code. That behavior is in the printer's own print start routine. So, the choice of slicer should not be a factor. Do you by chance have bed mesh disabled on the printer touch screen? That might change how jobs are done via screen vs via network.

The long temp soak actually recalls one other aspect of piezo sensors. They generate a voltage during temperature changes. That pyroelectric charge goes away after temperature change stops. A long pre-soak before print might let that voltage dissipate before nozzle probing. In contrast, heating up a bed from room temp and staring nozzle probing immediately after warm up might be doing so with a pyroelectric charge still present in the piezo elements. Don't know if that effect is significant in our Plus4's

guykuo, 2025-02-22:
Arrangement of foam and enclosing H-structure of printer piezo sensors may be worsening piezo sensor usability at high bed temperatures. Notice how the damping foam is large and applied to create a blanket trapping heat from bed at piezo sensor. Prior repRap work has shown that thermally insulation of piezo sensor can benefit usability at higher be temperatures. The Qidi build arrangement does exactly the opposite. Instead, a large foam piece traps heat at the sensors and the H-enclosure reduces airflow to equalize sensor temp with chamber.

[image]

nighteagle1974, 2025-02-22:
Hello, so i use an USB-Stick. I don't have change any setting on the printer except the language.
If i start the Bed-Leveling in the Tools-Menue it use the nozzle to touch the bed in the routine and it looks fine.

If i preheat the cahmber on 55 Degree and also the bed to 90 Degree and make an Bed-levelling over the Printers Display all looks fine - the nozzle touch the bed but not bending or something.

So in this case the temperature can not be the problem or is this random problem? Now i have good luck? I print now the second pass of Flow-Rate Calibration of my ABS Filament.

nighteagle1974, 2025-02-22:
I have checked the piezos sensors - i wrote above - i have resolder the PCB's with lead-solder. I also have set the foam a little bit higher and press it on all sides to the piezos.

nighteagle1974, 2025-02-22:
> Start of print includes nozzle touch (bed moves up to touch nozzle) to find z-offset. Bed mesh does not necessarily need to nozzle touch
> 
> [image]

JLRGURU, 2025-02-23:
Same mine with miltiple ABS or ASA. Always whe I use the printer with heated chamber ON

rsgq, 2025-02-27:
Can @QIDITECH respond to this? Your 800$ machine is destroying my print bed if I print anything else than PLA and your last comment on this was [3 months ago](https://github.com/QIDITECH/QIDI_PLUS4/issues/40#issuecomment-2499334691). It would be great if there was some fix or official guidance (maybe based on some of the suggestions in this thread) that is not "Well maybe replacing the sensors with the exact same sensors could work?"

nighteagle1974, 2025-02-27:
I have contact to Qidi-Support and they send me new Piezo-Sensors and PEI-Plate. I have only send an video of my malfunction.
But i think the problem is not so extremly as the beginning. I have change some things:

1. I change from Qidi-Slicer to Orca-Slicer (Because if i use Qidi-Slicer they preheat the cahmber before bed-leveling and Orca does it opposite)
2. I preheat the Chamber only to 38 Degree
3. I make an bed leveling by an cold chamber only bed to 90 Degree
4. i demount all Piezo-PCB - not the Piezos itself!!! Only the five PCBc - i resolder each part on it with lead-solder - because the PCBs soldered lead-free and lead-free soldering may cause issue on high temperatures - lost connections on the parts ans so on - many studies on this issues since years.
5. I isolate the piezos again with an one side glue foam little bit thicker around 5mm and press it complete over the Piezos.

So in my case i can see if i make bed leveling at 38 Degree Chamber and 90 Degree bed it move the nozzle to the bed but it don't touch it to much - i can not see any movement of the Printerhead.
If i preheat chamber to 65 Degree i can see the Nozzle touch the bed and the printhead is moving a little bit up. I have make a Video here:
https://waehlbar.com/download/ABL_withHeatedChamber.mp4

So in this case i preheat only to 38 Degree the chamber then start print - so the print beginns with bed-leveling in my case on every print - i don't have deactivated it. Then after leveling the printer begin heat up chamber and begin the print - during printing the chamber heat up from 38 Degree to 55 Degree asap  and so i can print all ABS fine without crashing.

I think the best idea to solve this issue is first isolate Piezos with foam and look if the Bed-Leveling from the Display únder Tools shows no touching the bed to strong means no moving of printhead up and down. Then use Orca so the preheat is after leveling.

robmue159, 2025-02-27:
I have the same issue.
When the printer is already hot, the piezos dont work. If I start with a cold printer and send the print via Orca, everything is fine.
Its like shown here: https://www.youtube.com/watch?v=OQIc6ORcJ-A (not my video)

Qidi send me new piezos with boards (and new PEI and nozzle). But they have the same issue.

Hoefully there is a solution for this. Maybe Qidi could sponsor some Beacons...

MJB500, 2025-02-28:
> I have contact to Qidi-Support and they send me new Piezo-Sensors and PEI-Plate. I have only send an video of my malfunction. But i think the problem is not so extremly as the beginning. I have [...]

I have run into the same situation. As long as the chamber isn't above 40c (bed can be at 100c), the print will run fine and not drag the nozzle. I can tell it will be ok during the first Z offset touch, because it doesn't lift the hot end/flex the rails. 

As an aside, I have turned up the chamber heater power to 60% (up from 40%). If the chamber is warm-ish, it only takes between 1-2 minutes to get it up to 55c. I might try going to 70% power, but right now it is printing fine as it is.

I do not think that we should have to do this. Qidi should replace the sensors with ones that can tolerate a heat soak. If enough of us keep pushing, maybe that will eventually happen.

rsgq, 2025-02-28:
> I have turned up the chamber heater power to 60%

How? And are you sure this won't melt the SSR? (Man this printer is a mess... 😆 )

MJB500, 2025-02-28:
> > I have turned up the chamber heater power to 60%
> 
> How? And are you sure this won't melt the SSR? (Man this printer is a mess... 😆 )

The power setting is in the printer.cfg file inside of the Fluidd software. 

I have the latest SSR and have run the machine more than 30 hours with the heater set to 60%. The original setting pre-SSR issue was 70%. It works for me, but make changes at your own risk. 

With this setting, it heats up fast enough that I am not seeing any warping from lack of pre-heating. Again, it is not ideal, but it is working in my case.

rsgq, 2025-03-01:
> As far as "this printer being a mess", I guess it depends on what you are comparing it to.

Yeah fair, but I can't imagine Bambu having problems like a fire hazard due to the SSR when they start shipping because they didn't properly test for 110V or the sensors failing and destroying the build plate on a lot of machines because they didn't properly test for chamber heating. (Neither of these I would call minor honestly)

There are some other minor issues: I also had problems with [the firmware update not working in the beginning](https://wiki.qidi3d.com/en/PLUS4/Manual/firmware-update#ui-update-failed-how-to-fix-it) (as the first update had to be done with a USB stick even though this was fixed in September, yet they're still shipping printers with firmware before that time). The wiki entry did not work, I had to debug for over a week with support until I finally could use the printer (didn't they advertise 15 minutes setup or something?), then there's the UI where the gcode images take an eternity to load and you can't press anything while they're loading and a few other annoyances like that.

And of course the printer's amazing (e.g. the PLA print quality out of the box was stunning) if you compare it to the likes of your ANET or my last one (7 year old Anycubic). But this is not the standard of the competition anymore. Bambu is. And I think in that respect Qidi seriously needs to reevaluate the focus they have on testing before shipping and also both parts and firmware quality. The only reason not to buy a Bambu over a Qidi for me right now is that Bambu is so closed source. And that's not that important to a lot of people.

MJB500, 2025-03-01:
> > As far as "this printer being a mess", I guess it depends on what you are comparing it to.
> 
> Yeah fair, but I can't imagine Bambu having problems like a fire hazard due to the SSR when they start shipping because they didn't properly test for 110V or the sensors failing and destroying the build plate on a lot of machines because they didn't properly test for chamber heating. (Neither of these I would call minor honestly)
> 
> There are some other minor issues: I also had problems with [the firmware update not working in the beginning](https://wiki.qidi3d.com/en/PLUS4/Manual/firmware-update#ui-update-failed-how-to-fix-it) (as the first update had to be done with a USB stick even though this was fixed in September, yet they're still shipping printers with firmware from before that time). The wiki entry did not work, I had to debug for over a week with support until I finally could use the printer (didn't they advertise 15 minutes setup or something?), then there's the UI where the gcode images take an eternity to load and you can't press anything while they're loading and a few other annoyances like that.
> 
> And of course the printer's amazing (e.g. the PLA print quality out of the box was stunning) if you compare it to the likes of your ANET or my last one (7 year old Anycubic). But this is not the standard of the competition anymore. Bambu is. And I think in that respect Qidi seriously needs to reevaluate the focus they have on testing before shipping and also both parts and firmware quality. The only reason not to buy a Bambu over a Qidi for me right now is that Bambu is so closed source. And that's not that important to a lot of people.

It just sounds like you need to buy a Bambu printer. You are also not comparing price points. The Bambu is significantly more expensive, smaller, and unheated. If that is what you need and want a "push a button and go" regardless of cost, then that is the best choice. I am not making excuses for Qidi. I get frustrated too. I just am willing to sort things out as the machine does what I need it to do, for the cost I am willing to pay. If Bambu had a heated printer this size with the same nozzle temp range for anywhere close to the same cost, I would have ordered that. They don't, so I didn't.

By the time I ordered my machine, the SSR was resolved and updated. So, not a real concern for me. The chamber heating, while disappointing, works well enough on my machine as it has been revised. Now that I know how to make it work, it is fine. The damaged build plate sucked, but they are sending a replacement. 

Again, this is not a "beginner" machine, in my opinion. Even a Bambu printer is not 100% foolproof, but it is close. For people with the money and no patients, that just want to print. They are the go-to. Send back the Plus 4 and move on. Sounds like you are done dealing with it anyway....

rsgq, 2025-03-01:
> You are also not comparing price points

true

> this is not a "beginner" machine

I can also agree with that. But I also think that printers should be reliable and safe though, no matter the experience of the customer.

> Send back the Plus 4 and move on.

I might have somewhat little patience and little understanding for mediocre quality control, but I have an even smaller understanding for closed down systems that spy on and try to control the customers. So I will stick with Qidi. I just want them to be even better than they already are.

MJB500, 2025-03-01:
Thanks for taking my comments in the spirit they were intended. 

I had done a fair amount of research on printers before I made my purchase. If the SSR problem had not been addressed at all, I would not have purchased it. If the chamber heater hadn't already be revised by the time I was looking at them, I also would have passed. I agree that neither of these issues is acceptable, beginner or expert owner. I was much later to the party than a lot of people. I benefitted from their suffering. 

I have run several ASA prints today. Now that I know not to heat soak the machine before hand, it has been running fine. No plate gouging. I just let it get down to about 40c in the chamber before running the next print. That seems to work for me. 

I hope you get yours sorted out now that it seems like you are going to keep it. I am happy to help if I can.

nighteagle1974, 2025-03-01:
Hello again,

so for me i have found out that a good piezo sensor disc can handle temperatures up to 90 Degree without compensation. The cheap ones for few cents have problems above 55 to 60 Degree. This means we have found the problem  -the discs are in bad quality. And this propblem is also on other printers. Bambu Lab have also some problems with this - and the PCBs and the Sensors looks identical to the from Qidi.
https://wiki.bambulab.com/en/x1/maintenance/replace-the-heatbed-force-sensor

So i think it is a problem of the right manufacteur for high quality ceramic piezo sensors - there are some posts about this probblems also from RepRap. It is not the printer itself - the quality of the sensors differs. Thats the reason many people have no problems with it - because it works. So i look for an high quality sensor and in the meantime Qidi send me new ones - i will test this new one and compare to the old ones and my sensors i want to order from a manufactuer - price is around 1 Euro instead of 0,06 Eur the cheap ones from toys etc.
The other problems with smoking etc. is only to 110V customers - in germany we have 240 V.  The Display is a little bit slow - and i want to deactivate preshow of the files - it takes to long to render the image. Sometimes it failed and shows broken pixel images.

But in all this - i read often on BambuLab other problems - like AMS and so on... so it looks like all have some problems - but not to much as in the past aroud 2016 my first trash-printer was a 3D System CubePro Trihead... a completely shitti machine--- cost me around 4000 € closed filaments - Aluminium-Nozzle and so on... and now i have a 799 Eur machines an it makes fantastic prints....

In the end i will buy a second one of this printer.

WoMec, 2025-03-08:
Hi.
Like many of you, I received a brand-new set of sensors. During the replacement, I found that 3 out of 4 sensors were really damaged. Three sensors were broken off the metal plate, and one sensor had a broken cable.

Additionally, I noticed that the leveling screws were really tight. So after the replacement, I left the leveling screws slightly loose to allow for more sensitivity in the sensors.

After that, I achieved a quite solid and stable Z-offset calibration across a range of temperatures—the highest tested was 90°C on the bed and 50°C in the chamber.

To all of you who are still experiencing this issue, double-check your sensors for damage and try loosening the leveling screws a little.

[image]

[image]

[image]

[image]

MJB500, 2025-03-08:
> Hi. Like many of you, I received a brand-new set of sensors. During the replacement, I found that 3 out of 4 sensors were really damaged. Three sensors were broken off the metal plate, and one [...]

Thank you for taking the time to put this write up together. I think it will be helpful for others having this issue. 

I received new sensors, but have not installed them yet. After seeing this, I will probably do it sooner than later.

Unkleskunky, 2025-03-08:
Hi MJB500,
I've been observing your issue from the sideline. I am curious, did you
loosen the leveling screws and then perform the calibration?
Mine are especially tight on one corner at least.I had an impact induced
issue and when releveling it seemed to be at its limit.
I would like to back them off and recalibrate and leave room for
adjustment, and for the benefit you mention.

Can you bulletpoint how I might go about doing this?

Thanks

US

MJB500, 2025-03-08:
> Hi MJB500,
> I've been observing your issue from the sideline. I am curious, did you
> loosen the leveling screws and then perform the calibration?
> Mine are especially tight on one corner at least.I had [...]

It is [WoMec](https://github.com/WoMec) you need to ask this question of.

WoMec, 2025-03-09:
@MJB500 - I'll test higher temperatures soon and post the results here.

@Unkleskunky - When assembling the heatbed after installing the sensors, I screwed the adjustment screws/wheels to about 50% of their original position. I also did not install the locking bolts. Now, I have approximately 3mm of the screw sticking out from the leveling bolt/wheel.

After this, I ran the platform reset from the printer's LCD (Tools -> Platform Reset). During this procedure, you manually level the bed using the four screws/wheels in the corners and a piece of paper

WoMec, 2025-03-09:
@Unkleskunky 
From the factory, there is an additional nut, a spring washer, and a regular washer under each leveling wheel in the corners. If you want to manually level the bed, you should loosen them before turning the leveling wheels. I removed those.

WoMec, 2025-03-10:
> [@WoMec](https://github.com/WoMec),
> 
> Thankyou, I am assuming that removing the additional components has had no
> ill effect?
> 
> Thanks again,
> 
> US
> […](#)

Apparently not :)

invertedflight, 2025-03-11:
Just found this thread and the prior one mentioned.  I print almost exclusively with ASA.
I have three Qidi Plus 4 printers and I have been complaining to Qidi support about the Z offset problem since mid November 2024.
At this point I have hundreds of emails with them, have sent probably a hundred photos and a dozen videos documenting Z offset and bed leveling not working properly on all three printers.

Qidi never once acknowledged to me that there was a design problem with their printers.  Most of their responses were to blame me in some way or suggest that something about what I'm doing is causing the problems.  Nothing they have suggested has fixed or even improved anything.  Some of their suggestions were just silly - like setting my temperatures lower than I need them to be.  They did send me one set of piezo sensors to swap out and I haven't done that yet, mostly because I suspect it will be a great waste of time just like everything else they've thrown at me.

Needless to say, this has been absolutely infuriating and continues to cause me problems on a daily basis.

I'm glad to have found this thread and I hope to see some solutions come out of it.  Until then, I'll just keep hovering over my printers like a hawk at every print start to frantically adjust the Z to the proper level.  And the pile of damaged build plates keeps getting higher...

MJB500, 2025-03-11:
> Just found this thread and the prior one mentioned. I print almost exclusively with ASA. I have three Qidi Plus 4 printers and I have been complaining to Qidi support about the Z offset problem [...]

Have you tried only preheating to 40C or below (using the heated bed only) and then starting the print? With my machine, it works every time this way with ASA. I have also increased the chamber heater power factor to .6 from .4 (you need the latest SSR, but this speeds up warming a lot). The warmup time from 40c to 55cc is only 1-2 minutes like this. I have not had warping or print failure using this method. I know this is not ideal or the "way it should work", but it allows you to get work done until something else can be found as a fix. You will know it will grind if during the nozzle bed touch (offset check) the print head/rails flex up. It means the sensors are heat soaked. You do not need to wait for the print to start to see if the machine is in "carving mode".

invertedflight, 2025-03-11:
> Have you tried only preheating to 40C or below (using the heated bed only) and then starting the print? With my machine, it works every time this way with ASA. I have also increased the chamber [...]

40 to 55c in 1-2 minutes?  We must own different printers.  I have my chamber heater power set to 1.0 and I'm pretty sure it can't heat anywhere near that fast.

In addition to the Z offset, I also find that bed leveling doesn't work properly.  So I'm hesitant to have the printer changing temperature during the first layer, likely making leveling even worse.  I've had the chamber and plate much cooler for some rare PLA printing and it seemed to work ok for that so I agree the problem seems to be related to temperature.  For a while I did try to run the preprint calibration at lower ASA plate temperatures (low meaning 90c) and before the chamber could get warm but I remember still having problems.

I basically have to babysit the entire first layer and make multiple adjustments to the Z offset as it prints different parts of the plate (usually I'm doing 4-12 parts in one print).

I often print serially and often at 60 or 65c chamber temp.  Even if your idea of doing calibrations on a cold printer worked, I don't want to wait around while the entire printer cools down just so I can start the next print.

MJB500, 2025-03-11:
> 40 to 55c in 1-2 minutes? We must own different printers. I have my chamber heater power set to 1.0 and I'm pretty sure it can't heat anywhere near that fast.
>
> In addition to the Z offset, I also [...]

Can you post a pic of your bed mesh? If you adjusting it on the fly for different parts of the bed, it has to be really far off.

Cooling the chamber (not the bed) down to ~40c from 60c doesn't take that long. Open the lid and door. It cannot take as much time as you having to stand in front of the machine(s) adjusting it does. If you turn the bed temp to 10c and let the chamber warm up to 40c, how long does it take the chamber heater to reach 60c? If it takes more than 3 minutes, something is really wrong.

It is the chamber temp, not the bed temp, that is heat soaking the sensors (in my experience, anyway). They just need to get below a certain point to read correctly again. When you do the initial Z offset touch, does it lift the print head/flex the rails? If so, the sensors are not reading correctly. 

I know that this is not the way it should work and am not excusing Qidi for not having a proper fix. I am, however, trying to help you make the most of a bad situation. Unless new sensors are made available (maybe ones from an X1E), you are going to need a workaround. Your method sounds very labor intensive.

invertedflight, 2025-03-11:
> Can you post a pic of your bed mesh? If you adjusting it on the fly for different parts of the bed, it has to be really far off.
>
> Cooling the chamber (not the bed) down to ~40c from 60c doesn't take [...]

My bed meshes (I have three printers) are all fairly flat.  Typical range is 0.15 to 0.25mm
When I can get the bed super flat I have less problems.  Also keep in mind that I'm using the entire bed, and for my prints I need a continuous first layer (it is not acceptable to have gaps between lines), and I'm using either a smooth or epoxy finish plate (less forgiving than textured PEI).

What I have observed is that the printer does the opposite of what you would expect: where the bed mesh has a higher area, the printer puts down thinner lines.  It is like the printer is overcompensating and raising the nozzle way too high in the high area.
I have seen this happen to an extreme where I've had a spec of plastic under the print bed - I see the lump in the bed mesh in fluidd, and when the printer prints over that specific area I get very thin lines barely pressed to the bed and too thin to join to each other.
This happens on all three printers.

I take it for granted that I have to stand at the printer and adjust the Z during the first layer, so adding extra time for the cool down and warm up just doesn't help me.
I haven't timed anything lately, but I remember back with the chamber power set to 0.4 it would be nearly 20 minutes from print start to when the chamber finally reached temperature.  I would guess at 1.0 power going from 40 to 65 with the plate hot takes 5-10 minutes depending on the shop temperature.  If the printer frame is already warm from a previous job it'll go faster.

I haven't observed any rails flexing but I'll pay closer attention - this is immediately after the print job is started and the extruder runs to the center of the bed and does the touch off routine?  Seems like it just touches off a lot with maybe the tiniest barely perceptible movement of the printhead as the nozzle touches.  Mine cycle many many times trying to meet the tolerance requirement here and I should mention that I tried implementing this: https://github.com/qidi-community/Plus4-Wiki/tree/main/content/more-accurate-bed-meshing

Yes, this is labor intensive and maddening (like when I walk away for just a moment and that is when the print starts and damages my plate due to the Z problem).  I need the output on a daily basis though so I've devoted some time to experimenting with fixes but most days I just need to get the prints started and going.  This isn't the old days where you could get a print started in a few minutes.

MJB500, 2025-03-11:
Yes, when the nozzle is doing the first touch off in the center of the bed is when you would see the print head/rails flex. This is when I knew it would gouge the print plate. The sensors are no longer responding correctly when this happens (from heat soak I believe).

My bed mess (for the full plate) is around .15. That should be fine. I am surprised you are having to manually adjust per region of the bed. 

5-10 minutes to reach 65 from 40 at 1.0 power factor sounds crazy. Is Fluidd reporting 100%? Are you on 220v or 110v?

Seems like you could manually level the bed and then switch off bed leveling. For your case, that might be better than what you are having to do now.

invertedflight, 2025-03-11:
Well the shop is about 10 degrees warmer now than it has been recently during the winter.  So I'm sure my printers are warming up much faster.  I do remember timing the print start at near 20 minutes a few months ago when Qidi had the default chamber power to 0.4
I'll try to actually time it at some point, but in reality the number of minutes really isn't the issue here.
In order for me to tolerate longer heating and print start times, I would need to get these printers working like my Bambu printers where I can hit print and walk away.  With my Plus 4 printers I'm nowhere near that level of trust.  Until then, it is easier for me to turn on the heaters for each printer when I walk into the shop, then when I'm ready to send print jobs things startup that much faster while I'm waiting to guard the Z setting.  To make things slightly more efficient, I can walk between about five printers during first layer to check the Z.  With any luck, all the printers will be done with first layer after 20 minutes then I won't need to touch them until later that night when I start my overnight jobs.

nighteagle1974, 2025-03-18:
So i have printed two spools ABS and it works now with the new Sensors. I use the Orca Slicer and now if i want to print PAHT-CF with over 300 Degree but the print don't start. So i thought the Printer can up to 370 Degree - but now it wrote the Error out of Temp Range Check GCode. Then i have check Orca-Slicer - they wrote under the filament settings temperature Range must be over 190 and under 300 Degree... so i can not print over 300 Degree? It is Printer related or Orca-Problem? In the printer Profile of Orca there is not any setting for  temperatures... i search this issue since 3 hours and nobody have this problem before... i wonder i'm the only one use Orca and Plus4 to print high temp PAHT ???

[image]

RaphaelVogel, 2025-03-18:
Why isn't there an official statement from Qidi about this issue? I just wanted to buy one, but looking at this.......

MJB500, 2025-03-18:
> So i have printed two spools ABS and it works now with the new Sensors. I use the Orca Slicer and now if i want to print PAHT-CF with over 300 Degree but the print don't start. So i thought the [...]

Did you try using QidiStudio slicer? Did you pick a Plus 4 in Orca? It might be the slicer, not the machine.

Alex00101, 2025-03-20:
> Why isn't there an official statement from Qidi about this issue? I just wanted to buy one, but looking at this.......

Plus 1 on this.... I was just about to hit "Complete Order" maybe 3 weeks ago, the magnitude of this problem and the fact that there is still no response yet alone the solution from Qidi makes me feel glad I did not, and perhaps never will.

invertedflight, 2025-03-20:
> Why isn't there an official statement from Qidi about this issue? I just wanted to buy one, but looking at this.......

I practically begged them to affirmatively acknowledge the issue back when I was sending them email after email showing my three Plus 4 printers screwing up the Z offset.  They just wouldn't budge, and kept suggesting that it was something about my settings like I shouldn't be setting the chamber temp above 50c.  They kept requesting proof, like videos, and I took the time to make several videos of the complete start sequence and first layer but then they never acknowledged them so that just ended up being a huge waste of time.  At some point it just feels like their goal is to wear me down.  So maddening.

JLRGURU, 2025-03-20:
> Did you try using QidiStudio slicer? Did you pick a Plus 4 in Orca? It might be the slicer, not the machine.

check the "printer.cfg" by SSH at the section **[extruder]**

[extruder]
step_pin:PB9
dir_pin:PB8
enable_pin:!PC15
rotation_distance: 53.7  #22.6789511 Bondtech 5mm Drive Gears
gear_ratio: 1517:170
microsteps: 16
full_steps_per_rotation: 200 #200 for 1.8 degree, 400 for 0.9 degree
nozzle_diameter: 0.400
filament_diameter: 1.75
min_temp: 0
**max_temp: 380**  <-------------------------------------------------------------------
min_extrude_temp: 175
smooth_time: 0.000001
heater_pin:PB3
sensor_type:MAX6675
sensor_pin:PB12
spi_speed: 100000
spi_software_sclk_pin:PB13
spi_software_mosi_pin:PA11
spi_software_miso_pin:PB14
max_power: 1

control : pid  
pid_Kp=33.555
pid_Ki=4.76
pid_Kd=59.141

pressure_advance: 0.032
pressure_advance_smooth_time: 0.03
max_extrude_cross_section:500
instantaneous_corner_velocity: 10.000
max_extrude_only_distance: 1000.0
max_extrude_only_velocity:5000
max_extrude_only_accel:5000
step_pulse_duration:0.000002_

nighteagle1974, 2025-03-20:
Thank you - it looks like is a slicer-problem. I have use a preset PAHT and in this the temperaures range go to 350 Degree. Have print and all is fine. But if i use an other like PLA Preset and change temperature to 310 Degree then the problem appear. Looks like there is a hidden setting for each Filament. But now i use PAHT Preset and go over 300 Degree.

JLRGURU, 2025-03-20:
> Thank you - it looks like is a slicer-problem. I have use a preset PAHT and in this the temperaures range go to 350 Degree. Have print and all is fine. But if i use an other like PLA Preset and [...]

Great! Thank you for sharing.....

nighteagle1974, 2025-03-20:
So for the Main-Problem of the Z-Piezo-Sensors - i found out with the new Sensors it is not the Sensor itself or one of the four that have a temperaure-problem. Because these piezos are stable to 90 Degree and high end up to 150 Degree.
The little Piezos are glued to the metall-sheet. And this is not the best idea to assemble two different metal materials together under this hot environment. I found a piezo that was loosen from the metall-sheet - was only pressed with the foam to the metal-sheet. The glue is not good and also the assemble work of the people that make the glue on it are not the best. I can say here we need a better idea to mount the piezos to this metal-sheet. But you can not screw it or amke housing - because it change the bending abiility of the sensor and metal-sheet. My idea is to mill a 1mm thick gfk-plate and mount it to the metall-sheet and between it is the piezo. I mill the part so that the sensor and it cables fitt 100% in it - and hope the bending change not to much. I will give it a try.

[image]

nighteagle1974, 2025-03-20:
I have now a simple solution - but little bit tricky. Need a Solder-Gun around 100W and solder it on the metall-sheet. But first you need to clean all up and also sand the surface of the metal-sheet - and then make solder on it - need miniutes to make a fine solder-surface. Then clean the piezo and solder to the metall-sheet only on the brass part. But in the end i make mistake to resolder the cable - don't do it - or do it with less temp. and quick. I have burned the fine metall-surface on the ceramic by solder the red wire.... But i have another 3 around to test this out.

[image]

DavidDugas, 2025-03-21:
Those of you that received new sensors...was there any discernable difference from the originals?  Qidi sent me new sensors but they look exactly like the original ones and the PCBs are identical in appearance and rev level to originals.

did44, 2025-03-21:
> I have now a simple solution - but little bit tricky. Need a Solder-Gun around 100W and solder it on the metall-sheet. But first you need to clean all up and also sand the surface of the metal-sheet [...]

Hello, Does the sensor welded to the plate work better?

sonny486, 2025-04-02:
Is @QIDITECH ever going to update the firmware to resolve this issue?

nemith, 2025-04-02:
unfortunately there is no way to troubleshoot the hardware signals.  Qidi sends you a test print and then sends you a lot of hardware to replace.

What is needed is some proper diagnostics.

Unkleskunky, 2025-04-03:
I have not yet encountered this but haven't done used any particularly high
temp mats, is this very common?

MJB500, 2025-04-03:
> I have not yet encountered this but haven't done used any particularly high
> temp mats, is this very common?
> […](#)

It mostly happens when heat soaking the chamber with the chamber heater on and the temp going above 40-45c before probing (in my case anyway). I posted the way I work around it, but some have replaced the bed sensors and not had further problems while other have issues after replacement. If you are not planning to print materials that need a high chamber temp, you most likely will not be affected.

Unkleskunky, 2025-04-03:
Thanks MJB500,

I do intend to do so. Is it possible to substitute a more effective sensor?
I am following your work around as well. I have also been following the
different technology sensor thread.
So is it correct to assume that the bed itself is changing in such a way
that the sensor cannot accommodate for? Or that the sensor is affected by
the heat? The reason I ask is I am curious if we could benefit from a plate
that may distribute heat differently? Maybe a a sandwich with copper, ( I
say this not from a position of materials knowledge but rather things I
recall reading) or another material?

Thanks for considering my musings, it is my hope that smarter people will
avail me of my ignorance on the subject. It is my understanding that Qidi'
Plus4 has a denser plate to tolerate the heat. Perhaps it needs some help
or it is acting to its detriment?

Thanks

Unkleskunky, 2025-04-03:
If unclear, when I refer to a plate "sandwich" I am contemplating one
taking advantage of the existing magnetic base plate. If such technology
were effective it could be utilized by other printers breaching the high
temp barrier.

MJB500, 2025-04-03:
> Thanks MJB500,
>
> I do intend to do so. Is it possible to substitute a more effective sensor?
> I am following your work around as well. I have also been following the
> different technology sensor [...]

The sensor is failing/becoming inaccurate at higher temperatures. It could be due to the glue they use coming loose from its mount. The soldering mod might work. Higher temp adhesives might also be an option.  I have not tried it as my workaround is fine for what I do at the moment. Currently there is no alternative sensor available that I am aware of.

Unkleskunky, 2025-04-03:
So temp sensors not z sensor, my mistake.

The soldering mod suggests no glue but direct thermal "Weld" with solder if
Im understanding correctly, what about a better quality sensor? Better
glue? Is it a standardized connection? Will code need editing?
I am hoping to fix before I have an issue.

MJB500, 2025-04-04:
> So temp sensors not z sensor, my mistake.
>
> The soldering mod suggests no glue but direct thermal "Weld" with solder if
> Im understanding correctly, what about a better quality sensor? Better
> glue? Is [...]

No, the bed (Z offset) sensor is what the issue is. The Z offset uses 4 sensors mounted under the bed to determine offset.

Yes, the suggestion from another user is to solder the sensor to the plate in place of using glue. I know only what that person posted above. A higher temperature glue might work as well. No one here (that I know of) has tried doing that. 

As I said previously, there are not upgraded sensor available for the Plus 4 (that I know of). Coding and connectors are not relevant since there are no alternative sensors that can be installed (that I know of). 

Qidi will send you replacements if you show that you are having the problem. There is no guarantee that they will 100% resolve the issue. Some users say they have, others say the problem persists. 

Some are installing Beacon or Cartographer sensors instead of relying on the bed sensors. This is a much more involved process, but details can be found with a little bit of searching.

Unkleskunky, 2025-04-04:
Aha,

So the other side of the z sensor, got it. Do we suppose there is thermal
paste underneath the sensor?
Perhaps mounts can could be sintered on and a less permanent physical
connection could be used. I think we must consider that the plate tolerance
itself is changing due to heat, which I had supposed the thicker plate
would deter. Maybe not.
I was supposing that these sensors might be substituted with something from
another manufacturer without this issue. Perhaps I can do some research
there.

"Some are installing Beacon or Cartographer sensors instead of relying on
the bed sensors." This is what I was referring to but did not recall the
technical description. Sounds pretty involved, I am hoping we can implement
something simpler first, though there were some other benefits that were
compelling, perhaps.

I am curious how common the issue is, too.

Thanks for the clarification.

TheLiTiG, 2025-04-22:
Hi, I have noticed the same issue on my unit. PLA and other materials have been printing fine but as soon as I started printing PC with a bed temp of 110c everything went south. My PEI Texture plate was printed on with such little distance that the fillament is deep in the texture. I also lost a Smooth Plate from QIDI in the process. Totally messed up and not repairable. I followed some advice to unscrew all the black screws completly and just slighte tightem them back up which apperently reduces the pressure on the bed so when it heats up it doesnt just expand upwards. 

I fixed that today and havnt printed with PC yet so unsure if it fixed the issue with all materials. I reached out to Qidi but havnt recieved any answer or spare parts to fix my sensors. For now PLA with a 55c bed temp works. 

If i remember to update I will. Its my first printer ever so I hope I can go back to trusting it. Really dont want to by a bambu lab printer.

[image]
[image]

[49 more replies]