QIDITECH/QIDI_PLUS4 #61: Camera is not working in X-Plus 4

Chas116, 2025-02-19:
My Camera is not working , Tech support was kind enough to give me a list of things to check and also installing the latest version. I performed all of these tasks, also checking the connection of the cable to the mother board.

but the camera still dose not work in fuildds using the QIDI Studio slicer.
I also down loaded the QIDI link app on my phone for both machines added to the app my Q1 Pro works fine on both my phone app and my desk top computer. what I have noticed is while running both printers when looking on Fluidd again the Q1 pro is working fine but the X-Plus 4 everything is working except for the camera and when its printing an error keeps popping up saying error taking timelape frame, when I close the error it pops up I believe after each layer.

on my phone when trying to connect or force connect it says no moonraker connection / please check moonraker status and / or refresh

I have a gut feeling that there is something missing or corrupted

please advise

also the tech support said they are sending me a new camera in the mail but I'm not convinced it's the issue

Grefixx, 2025-02-19:
Mine started same way, ended up with no connection to fluidd at all , only way is to get new emmc from qidi , and to get up and running meanwhile you have to downgrade fluidd version, still no access to link tho , only in lan.

Chas116, 2025-02-19:
What’s the emmc? The QIDI tech said they were sending me a new camera . Do
you have yours working now?

Grefixx, 2025-02-19:
Waiting for an answer from them , got it up and running for meanwhile by downgrading to a older version of fluidd. eMMC is the drive where all the system image is running on.

Chas116, 2025-02-19:
ok I new or had a feeling it was something in the softwear keep me in the
loop please, I just got an email from them letting me know the new camera
is on its way and a video on how to replace it , and I'm not going thru
that
not that I can't I don't have the time. It's crazy how much you have to
take apart just to change the camera. I'll take the wire off and plug it
into the new camera first to see if that's the issue before I proceed

QIDITECH, 2025-02-20:
I'm sorry, but you two have different problems. Please trust our after-sales team to solve the problem for you.