QIDITECH/QIDI_PLUS4 #68: LAN Connection Only kill network connection

eloo, 2025-03-10:
Hi,
since the update 1.60 it looks like the "LAN Connection Only" is broken. 
In my setup the connection to my printer is not possible anymore as soon i enable the "LAN Connection Only" mode. 

I am not able to ping the IP of the printer anymore nor can i access it for the UI or SSH. 

I was expecting the lan mode only "kills" the connection to internet services but it seems that it also kills the whole networking at all. 

Tested setup:

- DHCP with fixed IP
- Tested with WIFI
- Tested with LAN-Cable

Test scenario
- Started with "LAN Connection Only" disabled
- Ping the IP -> all OK
- Enable "LAN Connection Only"
- Ping not working anymore

Maybe the interface gets disabled or the ip is removed? 
But its hard to guess without access at the moment when "LAN Connection Only" is getting enabled.

QIDITECH, 2025-03-12:
Thank you for your feedback, we will optimize in the subsequent version.

sclead03, 2025-06-05:
Just found this searching for fixes as I just connected my printer for the first time. I have the same behavior, network works fine until "LAN Connection Only" is enabled, then the network is completely blocked. It would be great to see this fixed, as I'm relegated to USB drive only use until it is. This is a major selling point for Qidi over some of the competition who don't offer this option, sad to see it not functioning.

eloo, 2025-11-23:
@QIDITECH this is still not fixed with 1.7.3

its really getting ridiculous with you

looks really like we have wasted our money for your shit