QIDITECH/QIDI_PLUS4 #70: After updating to V1.6.0 only 31 makros avaliable Huperlaps no longer selectable

TinkerBarn, 2025-04-13:
I updated my PLUS 4 from 1.4.3 to 1.6.0 and saw in fluidd web interface settings that now are only 31 makros available. with 1.4.3 it was 57 macros.
Which macros are removed and why? Do we not longer need them?

Also I often used the HYPERLAPSE mode instead the TIMELAPES mode "Layer macro". I was able in 1.4.3 to set this a the default in the fluidd settings for timelapse. Now only "Layer macros" are available as a timelapse method.
How I can xhange it back to select between both methods?

Thanks

QIDITECH, 2025-04-14:
We didn't remove any macros, we just hid some macros that didn't run alone or could cause printer damage. Because users often use them incorrectly.
You can add the following configuration to the configuration file moonraker.conf and restart the printer to switch modes
mode: hyperlapse
hyperlapse_cycle: 10

[image]

TinkerBarn, 2025-04-14:
Thank you for your fast response.
I got this with macOS and can understand it.

But regarding the Switch between Timelapse Mode Layer macro and Hyperlapse I can not really understand, why this was removed?
Thank you for the advice, how I can switch to Hyperlapse with editing the moonraker.conf, but this bring not back the possibility to select the wanted method in fluidd settings for Timelapse.
Now it is fixed to Hyperlapse without the possibility to change the Intervall or switch back to Layer macro. Need to edit again the moonraker.conf
That could not be the user friendly option.