QIDITECH/QIDI_PLUS4 #72: Chamber heater

Meg1Print, 2025-04-16:
It works great, if you set the temperature of the bed to e.g. 90°C and the chamber to e.g. 55°C - no problems and all works save and sound.
But if you try to set the bed to 55°C and the chamber to 40°C - nothing happens. the heater bed heats up, but the chamber heater just starts the fan and thats it - no heating by the chamber heater at all - and so you never get the chamber temperature you wanted.
Even if you dare and deactivate the restrictions in the printer.cfg

_heat_with_heater_bed:True
heat_with_heater_bed_tem_add:25_

set to

_#heat_with_heater_bed:True
#heat_with_heater_bed_tem_add:25_

The most annoying part is that the chamber heater only starts to work **IF** the difference between bed temperature and chamber heater temperature is more than 25°C **AND** the bed temperature is set to 70°C. Below these settings the chamber heater does not work.
So these settings seem to be hardcoded in the firmware

JLRGURU, 2025-04-16:
_Even if you dare and deactivate the restrictions in the printer.cfg

heat_with_heater_bed:True
heat_with_heater_bed_tem_add:25

set to

#heat_with_heater_bed:True
#heat_with_heater_bed_tem_add:25_

I don't see any differences.....

Meg1Print, 2025-04-16:
deactivated by "#" ?

QIDITECH, 2025-04-17:
We are sorry that the temperature of the hot bed must be set at least 25° above the heat chamber.

Meg1Print, 2025-04-17:
hi Andy
please don't do that. it is not worth it to sell a bug for feature. it is proofed, that in firmware 1.4.3 it worked[Image]
but in  FW 1.6.0 it does not? there is no other explanation than it is a bug in the firmware.
especially when I look into the printer.cfg, disable the temp-difference there, but the printer does not react to that change and shows the same behavior like before the change. so why would a thinking developer do such a crap? there would be no need to implement commands in the printer.cfg that are useless!

and - sorry to say that - in none of qidi's descriptions of the plus 4 is mentioned, that the chamber heater is only working above a specified temperature, but not below that temperature and that it is only working in special difference to the bed temperature.
I'm curious what the community on discord and reddit thinks about that....

martin