QIDITECH/QIDI_PLUS4 #73: Can't Level Bed on Plus 4

MyPlus48MyBambu, 2025-05-02:
Everything was great on our Plus 4 but then the motherboard went bad.

Replaced the motherboard, and now when we try to level the bed all we get is a "Movement beyond range of provisions.” error.

Tried all of the following with no results:

--- Reinstalled firmware ( V1.6.0  ).

--- Reinstalled Printer.cfg file

--- Replaced Auto Leveling Sensor

--- Replaced Adapter Plate

--- Reset to factory settings

--- Replaced all Heatbed Sensors

--- Reset Klipper

Any help or guidance on how to fix this would be **greatly appreciated**.

QIDITECH, 2025-05-04:
I'm very sorry. Based on the above information alone, I can't determine where the problem lies. Please confirm whether the wiring is correct and whether the printer moves normally. If everything is correct, please record a video and contact after-sales service.

MyPlus48MyBambu, 2025-05-04:
All wiring is correct and printer moves normally.

Sent several videos to after sales service with no results.

QIDITECH, 2025-05-12:
I'm very sorry for not replying to you in time. Maybe you can contact other after-sales service email addresses, such as
karl@qd3dprinter.com