QIDITECH/QIDI_PLUS4 #74: Fluidd does not remember the interface settings.

user32rtu, 2025-05-07:
I am changing the Fluidd interface settings. After the re -opening of the Fluidd web interface, I see that my settings have not been preserved. How to make the interface settings remain?
Printer Qidi Plus 4. Firmware 1.6.

QIDITECH, 2025-05-12:
I'm very sorry for not replying to you in time. Could you take a screenshot and tell me which setting is not effective?

eloo, 2025-05-12:
hi,
i have also seen this..

in my case its related to the thermal presets

[image]

user32rtu, 2025-05-12:
Here are the settings that I fo

[image]
[image]
[image]

und.

QIDITECH, 2025-05-14:
I'm very sorry, but I can't reproduce this problem. All these functions are normal on my printer. I will keep an eye on this issue until it reappears and is resolved. I'm not sure how long it will take. If there are any other details worth referring to, please tell me.

eloo, 2025-05-14:
@QIDITECH you just need to change these settings.
e.g. add some thermal presets..

and then reboot the printer.. and you will see that your changes are gone

QIDITECH, 2025-05-20:
I'm very sorry. I tested on multiple printers, but still couldn't reproduce this problem.

ThorSc, 2025-05-25:
Hello, I have exact the same issue with my both Plus4.

linouxis9, 2025-05-25:
I can reproduce this issue too

alsmithson, 2026-03-25:
This was fixed for me in Firmware 1.7.3.