QIDITECH/QIDI_PLUS4 #84: after 1.7.,0 update, how do i load if i don't have a qidi box?

roadkingvrod, 2025-07-22:
It assumes I have a qidi box but I don't (yet).  But my load and replace filament options are gone and if i click on the box icon, it asks me to define what's in it.

How do i use with a single spool again until I buy one?

SilentShadow70, 2025-07-22:
I'm trying to figure out the same thing

dsieradzki, 2025-07-22:
The same, load filament button is disabled.

IMHO, when printer doesn't have connected qidi box it should use "old approach" on the screen.

It's quite series issue for people without qidi box. You can load filament from fluidd by macro but it's not solution, but only workaround.

dsieradzki, 2025-07-23:
Confirm, but....

This conditions make sense for qidi box, I sill strongly disagree with this when qidi box is not present. After homing the toolhead is in the middle of the bed, after set proper temp, the oozing is starting before toolhead is above poop shot. More over after start "load filemant" set temp is overridden by the printer. I suggest distinguish the two procedures with and without qidibox.

dgauche, 2025-07-24:
Yeah this is bad. @QIDITECH you need to fix this!

Meg1Print, 2025-07-25:
to make the "load" button work, you have to set the temp of the extruder to e.g. 240° - and the runout-sensor has to be free (no filament inside the toolhead) - and then same procedure as unloading - edit the filament, and the "load" button gets active

dsieradzki, 2025-07-25:
Ok, after following the procedure, I'm able to unload and load the filament. However, doing it manually without the box is simply not a convenient solution. I would prefer to distinguish between the procedures, but maybe that's just a personal preference.

That said, I have some concerns about the current unload and load filament procedure. After unloading, the screen shows a message that the task is complete. When you press "Complete", the toolhead moves away from the purge hole. If you do this too quickly, there is significant oozing onto the bed due to the high nozzle temperature (280°C) during unloading and the lack of cooldown on the purge pad after it.

If I then immediately click "Load Filament", the nozzle is heated back up to 280°C, and during the heat-up process, the printer starts homing. This leads to a mess on the bed because of the filament oozing from the hot nozzle.

TAGNG, 2025-07-26:
> The Hall value now determines the loading and unloading buttons. Please ensure that the nozzle is clear of any consumables before loading.

Please provide a step-by-step procedure, as certain filaments cannot be loaded or printed by loading them into the colour box. Why was this not considered when providing updated firmware??

TAGNG, 2025-07-26:
There should be a dedicated load/unload procedure if you do not choose to use a filament like PET GF or other brittle filaments. The same is also true of soft or flexible filaments

tetsujin-am, 2025-07-28:
1.7.0 アップデート後、印刷に入ること自体ができません。エラーコメントは「機械フィラメントは印刷ファイルと一致しない」となって動きません。BOXは使用していません。BOXのタブもオフになっています。どなたか解決策を教えてください。

tetsujin-am, 2025-07-28:
「機械フィラメントは印刷ファイルと一致しない」のエラーは解決しました。フィラメントの設定が1.7.0から新たに増えて造形機側でも素材設定をしないと動かないことがわかりました。

QuadrifoglioVerde, 2025-08-03:
There should be some warning like "do not update if you don't have box", reverting is not possible because after downgrade back to 1.6 printer cannot be calibrated :/

Gr3yW0lf2006, 2025-08-03:
you can "downgrade" by flashing the emmc complete and upgrade to 1.6.0. step by step by using a usb-stick
https://wiki.qidi3d.com/en/Memo/flash-emmc

vNate007, 2025-08-30:
> you can "downgrade" by flashing the emmc complete and upgrade to 1.6.0. step by step by using a usb-stick https://wiki.qidi3d.com/en/Memo/flash-emmc

Old thread but has anyone successfully completed a firmware downgrade using this method?