QIDITECH/QIDI_PLUS4 #95: Connection Issues over WiFi with new Firmware

Wanderwichtl, 2025-08-13:
Hi,
since I installed the newest update today, I have some connection issues.
The connection to the Qidi Plus4 is quite instable:  
```
ping 192.168.178.77 -t

Ping wird ausgeführt für 192.168.178.77 mit 32 Bytes Daten:
Antwort von 192.168.178.20: Zielhost nicht erreichbar.
Antwort von 192.168.178.20: Zielhost nicht erreichbar.
Antwort von 192.168.178.20: Zielhost nicht erreichbar.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=3641ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=651ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=755ms TTL=64
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=911ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=829ms TTL=64
Antwort von 192.168.178.77: Bytes=32 Zeit=403ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=749ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=919ms TTL=64
Antwort von 192.168.178.77: Bytes=32 Zeit=739ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=854ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=908ms TTL=64
Antwort von 192.168.178.77: Bytes=32 Zeit=382ms TTL=64
Antwort von 192.168.178.77: Bytes=32 Zeit=229ms TTL=64
Antwort von 192.168.178.77: Bytes=32 Zeit=238ms TTL=64
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Antwort von 192.168.178.77: Bytes=32 Zeit=782ms TTL=64
Antwort von 192.168.178.77: Bytes=32 Zeit=796ms TTL=64
```
Sometimes I do get a short connection in the orca slicer. I have restarted every part of the system (orca slicer, router, qidi).  
Maybe someone is experiencing something similar?

roadkingvrod, 2025-08-15:
I am experiencing the same thing.  Sometimes it connects, and sometimes it doesn't.  On 1.6x and 1.7.0, i had no problems with wifi.

SergioGiacomoSammartano, 2025-08-20:
Yeah, same issues here.

SergioGiacomoSammartano, 2025-08-20:
To be more precisely, I struggle to connect to moonraker:

[image]

DmitriARZ, 2025-10-06:
Доброго дня! Неужели ни кто данный вопрос так и не решил? Если решили подскажите  пожалуйста решение по данной проблеме.